chipsalliance/chisel #102: Call it SyncMem

schoeberl, 2016-02-22:
Rename the new memory from sequential to synchronous memory (e.g., SyncMem). That's the term that is used for memory chips. E.g., SSRAM is a synchronous static RAM, in SDRAM the S is for synchronous.

Basically there are two types of memories (or better memory interfaces) available: asynchronous and synchronous. The term sequential is not use for memories. And if your really want to, I would say both asynchronous and synchronous are 'sequential'.

ducky64, 2017-01-26:
Trying to clear our issues backlog.

Possible solutions are:
- have an alias for SyncMem -> SeqMem, where people could use either (but this would probably be really unclean and lead to wildly different styles)
- deprecate SeqMem in favor of SyncMem
- leave things as is

Making this more difficult: all memories are synchronous written, though (but some are combintaionally read)

aswaterman, 2017-01-26:
Sync vs. async isn't right, either.  Mem still has synchronous write ports (unlike asynchronous SRAMs, that use write strobes instead of clock edges).

aswaterman, 2017-01-26:
If the name is changed, the new name should actually clarify that the distinction is that read ports are synchronous instead of combinational.  SeqReadMem or SyncReadMem would be better than SeqMem or SyncMem.

ducky64, 2017-02-06:
Resolution:
- Rename SeqMem to SyncReadMem
- Keep Mem (the asynchronous read version) as Mem, under the reason that this is easier to use and probably what most people want.
- Deprecate old versions in `chisel3._`
- Keep old version without deprecation in compatibility layer `Chisel._`

aswaterman, 2017-02-06:
cool

schoeberl, 2017-02-07:
Fine with me.

Martin

PR #490 (2017-02-07): Rename SeqMem to SyncReadMem - fixes #102.
Retain un-deprecated SeqMem in compatibility mode, deprecate in chisel3.
Fixes #102.
closed 2017-02-07 by ucbjrl