chipsalliance/chisel #103: Build instructions

schoeberl, 2016-02-22:
What is the correct setting in build.sbt to use the local published version of Chisel 3? I tried to set it similar to Chisel 2:

libraryDependencies += "edu.berkeley.cs" %% "chisel" % "3.0-SNAPSHOT"

But got:
	Note: Unresolved dependencies path:
[warn] 		edu.berkeley.cs:chisel_2.11:3.0-SNAPSHOT

ucbjrl, 2016-02-22:
Check the �version� line in chisel3/build.sbt.

It�s currently:

    version := �3.0"

so your library dependencies should be:

    libraryDependencies += "edu.berkeley.cs" %% "chisel" % �3.0�

schoeberl, 2016-02-22:
Thanks! that fixed it. Maybe add this to the README.

BTW: You're having non-ASCII quotation marks (single and double) in your email, which end up as non-printable characters (in my view) in Mail and Firefox.

ucbjrl, 2016-04-06:
Closed by #133

closed 2016-04-06 by ucbjrl