chipsalliance/chisel #104: Chisel emits firrtl files with undeclared wires

chick, 2016-02-23:
Creating a circuit with an IO of the following form
```
class Packet extends Bundle {
  val header = UInt(width = 1)
}
class VectorPacketIO(n: Int) extends Bundle {
  val ins  = Vec(n, new DeqIO(new Packet()))
  val outs = Vec(n, new EnqIO(new Packet()))
}
```
coupled with the fact that DeqIO and EnqIO contain initialization code emits a .fir file that firrtl blows up with the following messages
```
[DEBUG] Starting High Form Check
Exception in thread "main" firrtl.passes.PassExceptions: 
firrtl.passes.CheckHighForm$UndeclaredReferenceException: VectorPacketIOUnitTester4043176957748550505.fir@8.4: [module BrokenVectorPacketModule] Reference T_7 is not declared.
firrtl.passes.CheckHighForm$UndeclaredReferenceException: VectorPacketIOUnitTester4043176957748550505.fir@9.4: [module BrokenVectorPacketModule] Reference T_15 is not declared.
```

chick, 2016-02-23:
I can fix the problem by removing the initialization code in DeqIO and EnqIO and moving it to a method, that the circuit must invoke but I am not sure if that is the correct approach.  This is a distillation of the Router example in the chisel-tutorials. Please advise

aswaterman, 2016-02-23:
@jackbackrack came up with the approach of emitting firrtl line by line so might have something to say about what to do about this.

aswaterman, 2016-02-23:
@chick I think there are a few problems here.  It's not well defined when it's legal to create Chisel nodes, so it's not even obvious whether this should be considered incorrect.  Regardless, this kind of issue clearly shouldn't manifest as an opaque FIRRTL error.

A possibly acceptable hack is to auto-declare Wires on first reference.  This requires the Chisel3 frontend be aware of FIRRTL scoping, which would be a significant departure--but possibly necessary.

sdtwigg, 2016-02-23:
It doesn't seem like that would solve the problem, even in this case, since the created logic would need to manipulate the IO, not an internal Wire. There is a huge catch-22 here: you need to finish initializing the Bundle to see what it will bind to (Wire, Register, Port, Memory, etc.) but then you can't run more initialization code once you leave the constructor! Even if you could somehow, figure out that EnqIO was going to become a port (although with Chisel3 supporting internally directioned wires, even this could be an unsafe assumption), this doesn't generalize to other datatypes that might end up in registers as then you'll truly have no way of figuring out whether it should be in a register or wire.

Chisel Bundles were never designed to support initialization logic. (In fact EnqIO and DeqIO shouldn't be in the standard library where other users will try to replicate this behavior.) This seems roughly akin to trying to hack on an init method that is always called when a c-struct is created. You could 'fix' this by just moving that logic into an init method and writing in the documentation for these types that init method should be called before use. (There are further generalizations that can be made with this init method approach although perhaps keep it simple for now....)

ducky64, 2016-02-23:
One question we may want to ask ourselves is whether we want to think of a Bundle as just a data structure, or whether it should be able to generate circuit elements. I think limiting Bundles to data only has a certain simplicity and elegance (as well as a more 1:1 mapping to FIRRTL). Perhaps initialization logic (or other "active" Bundle ideas) should be part of a higher abstraction layer.

A caveat that you need to call init seems like bad user experience - just one more thing the user has to remember to do. For this in particular, does it make sense to put the initialization in a companion object, which also nicely separates the Bundle definition (a ready/valid) from the initialization / behavioral semantics?

chick, 2016-02-23:
I think the issue is a strong case for a correct version (whatever that is) of DeqIO and EnqIO belonging in the standard library so people can see how to do it right.  And while there are some big issues here, I think we need to have a pragmatic fix for now so there can be a reasonable example for the tutorial.  I'd vote for moving the current initialization code  in DeqIO and EnqIO into init methods (there seems to be agreement that the extant code is wrong given current theory).  I have working local branch that does this and it seems to work.

ducky64, 2016-02-26:
Temporary fix in #109. Not a long-term substitute for figuring out how to do this **right**.

aswaterman, 2016-02-26:
I hope to spend some time thinking about this over the weekend.

chick, 2017-02-02:
This has largely been addressed by various API updates, and has other issues and PR associated with further cleanup of Deq Enq

commit 8974f749 (2017-02-03): Added vec IO tests for #104 (#480)

* Added vec IO tests for #104

* Added Vec test case for Reg of vecs

* Change Vec creation to check if gen is lit (and hence needs to be declared)
Fixes #104

* Fix tests (add IO())), Vec.fill()

* Fix deprecated usage.

* Add Binding IO() NPE fix so tests pass.

* Fix style - use space consistently.

* Fix style - use space consistently.
closed 2017-02-02 by chick