chipsalliance/chisel #115: Drowning in warnings in chisel3

chick, 2016-03-08:
Running GCD in chiselTests gives a bunch of errors of the following forms.
```
%Warning-STMTDLY: /var/folders/ls/0xl3wjy949b2b9j36yn36v3c0000gn/T/GCDTester2434714804424790707/GCDTester1526173926997725650.v:31: Unsupported: Ignoring delay on this delayed statement.
%Warning-WIDTH: /var/folders/ls/0xl3wjy949b2b9j36yn36v3c0000gn/T/GCDTester2434714804424790707/GCDTester1526173926997725650.v:27: Operator EQ expects 32 bits on the RHS, but RHS's CONST '1'h0' generates 1 bits.
```
For problems in the chisel-tutorials, there can be dozens or more of these lines and they dwarf the actual output.
Question:
Are these errors important, if so, can they be fixed.
Otherwise, can we suppress them, new chiselers will be unnerved and overwhelmed with this output

aswaterman, 2016-03-08:
They're benign.  Ultimately, the right thing to do is to emit Verilog that
doesn't generate warnings, but for now suppressing them should be OK.

aswaterman, 2016-05-04:
I believe @azidar and I have fixed these with recent FIRRTL PRs.

closed 2016-05-04 by aswaterman