chipsalliance/chisel #121: Assertions Fire on 1st rising clock edge

jackkoenig, 2016-03-16:
In rocket-chip we have a problem where lots of assertions fire on the first rising clock edge. While the generated FIRRTL and Verilog does have conditional statements predicated on reset not being asserted, in rocket-chip, we have pipelined resets where it takes a couple of cycles for reset to propagate from the top-level reset to each module. This leads to assertions firing because their local reset randomly initialized to zero, thus the assertion can fire even though reset has yet to be asserted.

Chisel 2 solves this issue by creating non-synthesizable 1 bit values that initialize to 0, and are set to 1 the first time the local reset is asserted. 
```scala
  always @(posedge clk) begin
`ifndef SYNTHESIS
// synthesis translate_off
  if(reset) T0 <= 1'b1;
  if(!T1 && T0 && !reset) begin
    $fwrite(32'h80000002, "ASSERTION FAILED: %s\n", "VoluntaryReleaseTracker accepted Release that wasn't voluntary!");                                                                                                                                                    
    $finish;
  end
// synthesis translate_on
`endif
```

There are a couple of ways to solve this in Chisel 3: We could add similar unsynthesizable magic to the FIRRTL Verilog emitter. This is problematic, however, because FIRRTL does not treat resets as special so this might be either specific to Chsiel 3 generated FIRRTL or use some inference that would probably be brittle. It could be enhanced by adding a "reset" field to printf and stop in FIRRTL.

The better solution in my opinion is in Chisel 3. We could propagate the top-level reset to each register, and include this reset in the condition for allowing any assertions to fire. This adds a restriction that global reset must be asserted for long enough that it overlaps with every local module reset, but this isn't all that bad of a restriction. It has the nice property of being synthesizable too.

ccelio, 2016-03-16:
I don't know what your future plans were for supporting assertions on FPGAs, but I'd recommend a solution that is synthesizable (when the designer desires, so it's not extra stuff on ASIC) and won't back us into a corner later.

ducky64, 2016-03-16:
Would it be possible to figure out a synthesizable way to make sure all resets are asserted at system startup? Otherwise, I feel you're playing a game of whack-a-mole (you may solve assertions, but what about printfs, or some other reset-based construct that we haven't invented yet?) rather than solving the actual problem.

I don't know enough about the rocket-chip architecture, but could (for example) the pipelined resets be ORed with the global reset? Though this may require extra hardware and affect performance (but don't know by how much).

Also about propagating top-level reset in Chisel3 core (I assume that's what you meant): are there designs with multiple reset domains? If so, how can that be handled in a fully general way?

aswaterman, 2016-03-16:
Conveniently, the current implementation actually is synthesizable.  The assumption is that the global reset also resets the synthesized assertion-reporting hardware.  As long as the global reset is held high for at least as long as the reset pipeline depth--a reasonable constraint--this will do the right thing.

It's actually only for simulation that the current approach is broken, since the assertion-reporting system (fwrite/finish) does not respect the global reset.  So, doing the Chisel2 thing might be a sensible solution.  For synthesis, you'd just drop T0 from the condition.

jackkoenig, 2016-03-16:
It's a good point that the current implementation is synthesizable, we just really need a way in simulation to mask out assertions until we come out of reset.

I think what @aswaterman said about propagating the top-level reset, adding it to the conditions on printfs and stops, but dropping this for synthesis makes the most sense. I don't know of any designs with more than _top-level_ reset domain, I believe if you wanted to add more you would have to put them in the top-level io bundle. It seems to me that still propagating the top-level Module's default reset signal is still okay, it just requires that in simulation that particular reset is asserted while the other custom resets are asserted until everything gets reset. Seems find to me.

aswaterman, 2016-03-16:
If you just emit the code like Chisel2 does, where everything is guarded by `ifndef SYNTHESIS, there's no need for annotations.

jackkoenig, 2016-03-16:
For software simulation sure, but if I want to write a FIRRTL pass that makes assertions work on the FPGA, I will want to transform printfs and stops and extract their information to the top-level. This extra condition doesn't need to be a part of that and could be dropped if it were annotated as such. Maybe this really doesn't matter. 

 For taping out though I agree with `ifndef SYNTHESIS guard is fine.

aswaterman, 2016-03-16:
I see what you mean

colinschmidt, 2016-03-16:
For what its worth, the current Hurricane/Raven chips have a top-level reset that brings up parts of the uncore to properly initialize some of the IP on the board via HTIF. So that IP has a second reset that occurs later.

jackkoenig, 2016-03-23:
Fixed by #124  and ucb-bar/firrtl#92

closed 2016-03-23 by jackkoenig