chipsalliance/chisel #125: Chisel3 compilation error on Travis

zhemao, 2016-03-25:
The error message starts at

https://travis-ci.org/ucb-bar/rocket-chip/jobs/118401486#L1382

The problem seems to be in the module "RegressionTest".

https://github.com/ucb-bar/groundtest/blob/master/src/main/scala/regression.scala#L399

What exactly is the issue here?

palmer-dabbelt, 2016-03-25:
The error message says

````
Starting Check Initialization

Fatal Uncaught Exception.

/home/travis/build/ucb-bar/rocket-chip/vsim/generated-src/Top.BroadcastRegressionTestConfig.fir@16774.4: [module RegressionTest]  Reference regressIOs[0].cache.resp.bits.addr is not fully initialized.
````

which seems it's got something to do with not having initialized wires.  Poking around the FIRRTL source code, I see

````
firrtl $ grep "Check Initialization" -R src/
src/main/scala/firrtl/passes/Checks.scala:   def name = "Check Initialization"
````

and looking at the source it kind of indicates that's what is going on here -- it looks like this walks through every signal and recursively attempts to ensure that every signal is wired up to something.

I don't see anything in RegressionTest that's actually initializing the outputs that this pass is complaining about (cache.resp.bits.addr, for example).  I think this is probably the problem.  It seems this shouldn't be too hard of a case to check in Chisel, so we can give users a better error message -- but I'm not going to do that.

Also, note that your stuff doesn't appear to work in Chisel 2: <https://travis-ci.org/ucb-bar/rocket-chip/jobs/118401486#L2509>.

zhemao, 2016-03-25:
> I don't see anything in RegressionTest that's actually initializing the outputs that this pass is complaining about (cache.resp.bits.addr, for example)

Okay, that was my fault. I forgot to connect cache.resp.bits. But I fixed that and now it fails verilog compilation. As for the chisel 2 failure, that shouldn't be happening either. It was working until recently.

zhemao, 2016-03-25:
The Chisel2 problem seems to be with ROM generation. When I take out the following line

    val data_beats = Vec.tabulate(tlDataBeats) {
      i => UInt(0x3001040 + i * 4, tlDataBits)
    }

And replace it with equivalent logic not using a ROM, it works in the emulator. Did something break emulator ROM generation?

zhemao, 2016-03-25:
Also, as for the disconnected wires thing. What if you had a decoupled or valid interface and you just want to disable it? Currently, the way we do it is to just tie valid to false and leave bits disconnected. But the way it is now, you will also need to set bits to some default value or get a fatal error. It seems unnecessarily burdensome to require the RTL writer to connect all wires even if they will never be used.

aswaterman, 2016-03-25:
We support the old chisel2 approach for disconnected module I/Os -- all
instance inputs and all module outputs are pre-initialized to random.

zhemao, 2016-03-25:
Then why the fatal exception?

zhemao, 2016-03-25:
So I've deliberately connected all the wires and worked around the ROM generation issue. But still not able to compile the regression tests with Chisel 3. I've pushed my changes to the branch uncore-fixup on rocket-chip.

aswaterman, 2016-03-25:
What's the Chisel 3 problem now?

aswaterman, 2016-03-25:
I need to look at the FIRRTL to answer that.

zhemao, 2016-03-25:
Jack is taking a look at it now. Seems to be an issue with name expansion.

aswaterman, 2016-03-29:
any resolution?

jackkoenig, 2016-03-29:
Not yet, I'll be looking at this after 4

aswaterman, 2016-04-14:
I believe this is resolved in @jackkoenig's yet-to-be-merged FIRRTL PR.

closed 2016-04-14 by aswaterman