chipsalliance/chisel #127: CheckHighForm$UndeclaredReferenceException

zhemao, 2016-03-31:
Getting another strange Travis failure.

https://travis-ci.org/ucb-bar/rocket-chip/jobs/119705552#L1264

What does this mean?

jackkoenig, 2016-03-31:
It means that in the file Top.SplitL2MetadataTestConfig.fir, in module
MetadataArray, there is a reference on line 5132 to "T_7854" that was not
declared.

It's an excellent question *why* this is happening, but probably one of the
following:  Chisel is emitting the reference without declaring it, FIRRTL
is dropping it somehow, or the logic detecting this error is just wrong.

We need to add a pass that makes it easier to see what temporary nodes come
from where, but if you look for that reference in the FIRRTL file and
verify it isn't declared as a wire or node, you can confirm it's a Chisel
bug of some sort.

zhemao, 2016-03-31:
It's not declared in the FIRRTL file.

zhemao, 2016-03-31:
The problem appears to be with the following code in MetadataArray.

    if (hasSplitMetadata) {                                                       
      val tag_arrs = List.fill(nWays){ SeqMem(nSets, UInt(width = metabits)) }    
      val tag_readout = Wire(Vec(nWays,rstVal.cloneType))                         
      val tags_vec = Wire(Vec.fill(nWays)(UInt(width = metabits)))                
      (0 until nWays).foreach { (i) =>                                            
        when (rst || (io.write.valid && wmask(i))) {                              
          tag_arrs(i).write(waddr, wdata)                                         
        }                                                                         
        tags_vec(i) := tag_arrs(i).read(io.read.bits.idx, io.read.valid && rmask(i))
      }                                                                           
      io.resp := io.resp.fromBits(tags_vec.toBits)                                
    }

One of those vectors is not getting generated correctly.

aswaterman, 2016-03-31:
You shouldn't wrap Vec.fill in a Wire

aswaterman, 2016-03-31:
In this case, you actually shouldn't use Vec.fill -- you should write
Wire(Vec(nWays,UInt(width = metabits)))

On Wed, Mar 30, 2016 at 8:55 PM, Andrew Waterman <waterman@eecs.berkeley.edu

zhemao, 2016-03-31:
Well. That fixed it. I wonder why it was passing before.

aswaterman, 2016-03-31:
Mysterious.

closed 2016-03-31 by aswaterman