chipsalliance/chisel #131: Problems with methods defined in Bundles

jackkoenig, 2016-04-05:
The code below:
```scala
class Foo(val bar: Int) extends Bundle {
  val num = SInt(bar)

  def sign: Bool = num < SInt(0)
}

class FooTester extends BasicTester {
  val a = new Foo(5)
  val b = new Foo(-3)
  val c = new Foo(2)

  stop()
}
```
Gives the following FIRRTL:
```
circuit FooTester : 
  module FooTester : 
    input clk : Clock
    input reset : UInt<1>
    output io : {}
    
    io is invalid
    node T_10 = eq(reset, UInt<1>("h00"))
    when T_10 :
      stop(clk, UInt<1>(1), 0)
      skip
    node sign = lt(asSInt(UInt<4>("h05")), asSInt(UInt<1>("h00")))
    node sign = lt(asSInt(UInt<3>("h05")), asSInt(UInt<1>("h00")))
    node sign = lt(asSInt(UInt<3>("h02")), asSInt(UInt<1>("h00")))
```
It seems there might be some kind of reflection issue. I'm also wondering if this might be an anti-pattern? Is it okay to define methods on classes that extend Bundle? It seems like it should be supported.

ccelio, 2016-04-05:
Defining methods in bundles is absolutely critical. I can't understate that enough.

jackkoenig, 2016-04-05:
Howie informed me that this is because of the limitations of reflection, if you have a method with no arguments, it looks the same as a regular val in the Bundle. So the Chisel pattern is to use a dummy argument, eg.
```scala
class Foo(val bar: Int) extends Bundle {
  val num = SInt(bar)

  def sign(dummy: Int = 0): Bool = num < SInt(0)                                                                                                                                                                                                                              
}
```
This works. It does lead to the annoying consequence that Foo.sign must be called with parentheses even though it doesn't have any side effects so Scala-style says it should be called without them.

ccelio, 2016-04-05:
Oh yah that's true in chisel2. But seems like we either need to support no arg methods or catch and error out. Generating the above code seems very dangerous.

aswaterman, 2016-04-05:
If we knew how to catch it, we'd already be generating the right code for this case :-)

colinschmidt, 2016-04-05:
Just another user of methods in bundles chiming in that this is a cool feature I would need to have working in chisel3 to finish my upgrade.

jackkoenig, 2016-04-05:
Well, guess this is resolved. A good candidate for the Chisel Cookbook

ducky64, 2016-04-05:
You mean adding dummy parameters as the "official" pattern? I think this is a temporary workaround at best. Macros may provide a better "eventually" solution, especially since we want to ditch reflection anyways.

jackkoenig, 2016-04-05:
Macros are the obvious "eventually" solution, but this dummy argument is hardly a "temporary workaround" since it's been present in Chisel code for years. Maybe we can fix this by beta, maybe not. Regardless I'll reopen so it stays on the agenda.

ucbjrl, 2016-04-05:
Paul Grebe used a "checkPort" method in Chisel2 defined in Bundle that was used to validate "ports". It could be overriden so "exotic" Bundle classes could provide their own mechanism for validating their ports and distinguishing them from non-port methods.

sdtwigg, 2016-04-05:
Java reflection (by intentional Scala design) cannot tell if a public field access is a member or a no-argument function. You have to use Scala reflection to determine this.

I played around with this at some point. Here is a starting point for determining the incantations of scala.reflect to get all val fields that are of subtype Data within a class.

https://github.com/sdtwigg/gama/blob/master/chiselfrontend/src/main/scala/frontend/implementation/Reflection.scala#L20

I'm curious if you end up finding a case that doesn't work. As of around Scala 2.11.4, the invocations were successful in distinguishing between vals and no-argument functions. The general logic was to filter for (public) vals that also had a public getter method. Note that any non-public vals will not be found.

ccelio, 2016-04-05:
> Note that any non-public vals will not be found.

I'm having a hard time imagining what a non-public val inside of a Bundle would even mean, so this seems a reasonable trade-off.

sdtwigg, 2016-04-05:
> Macros are the obvious "eventually" solution

Macros (defined as scala compile-time macros, likely macro annotations) are not necessarily the obvious solution as the current implementation of macros* has very weak access to the type system. Compile-time macros basically involve rewriting the AST enclosed by the macro invocation. Thus, you can only safely rely on 'seeing' the vals and defs defined in the 'body' of that macro invocation. That constrained local view would not work well for (assuming it is as I remember it) the Uncore, which creates Bundles of many wires by mixing in traits that each include a subset of the wires.
Similarly, it is non-trivial to see what the type of an AST node when running a macro (Basically, it is hard to see, at macro elaboration time, whether a val actually returns a Data or subtype of Data, see * again).

Runtime reflection, on the other hand, can much more reliably see everything like class inheritance hierarchy and members since the code is already fully elaborated. (Amusingly similar to reasons why doing certain things in a backend IR is easier than in during Chisel elaboration....)

PS: This all assumes you not willing to tolerate API changes. If you are willing to adjust the API for defining Bundles then you can simplify the problem sufficiently that macros can solve it (although you could then simplify such that you don't need macros either...). Also, if you are willing to write a full-blown scalac plugin, you could 'more clearly' solve these problems as such plugins have access to the typer details, etc.

* This is actually due to resolvability issues. Whitebox macros and macro annotations must run before the final typer phase as they are able to influence the types inferred on expressions or (for annotations) even what types are available to be inferred. Thus, you, in general, cannot rely on the typer having all types available by the time those are invoked, particularly for macro annotations that, iirc, run before the typer. (The documentation actually explicitly calls out trying to invoke the typer 'early' in these cases could cause an infinite loop if the classes contain references to each other.)
Of course, the reimagining of the scala macro system in scala.meta could change everything; however, that is still in early-ish alpha and they are still speccing out the ultimate capabilities.

sdtwigg, 2016-04-05:
> I'm having a hard time imagining what a non-public val inside of a Bundle would even mean, so this seems a reasonable trade-off.

```scala
class Pathological extends Bundle {
  private val myHiddenWire = Bool() // filtered out
  val myPublicWire = Bool()
}
```

jackkoenig, 2017-01-18:
This was fixed by #342

closed 2017-01-18 by jackkoenig