chipsalliance/chisel #135: Bits don't support BITwise operations

colinschmidt, 2016-04-09:
I am unable to logically or, and, xor, etc. two Bits together.
```
value | is not a member of Chisel.Bits
```
The definition of Bits has a comment in it
```
  // TODO: perhaps make this concrete?
  // Arguments for: self-checking code (can't do arithmetic on bits)
  // Arguments against: generates down to a FIRRTL UInt anyways
```
I would like to argue for having these operations. 
Having a datatype that can only have logical ops performed gives the designer useful feedback when they inadvertently perform arithmetic on it.

The argument in the comment against doesn't seem sensible to me. 
If chisel3 can help the designer make fewer mistakes why does it matter that the next stage of the compiler can't provide the same check.

ccelio, 2016-04-09:
In chisel2, bits currently aliases to uint. In the past they were fully separate, but we found zero utility from the type-checking. Instead it littered the code with toUInt casts and the designer just did whatever casting was required to make the system shut up. 

It does seem silly to not let you do logical operations on Bits. I guess my vote is to keep Bits and UInts identically, since in 4 years we've not seemed to find that all that problematic.

aswaterman, 2016-04-09:
I disagree with the premise that it's useful to have a type that supports logical ops but not arithmetic.  In what cases does this truly provide additional safety?  In my experience, it just results in a proliferation of casts to UInt.  Chisel2's use of the Bits constructor as an alias of UInt serves as a documentation hint, which serves most of the purpose.

But, assuming such a type really is desirable, it should be a different type, not a supertype of SInt and UInt.  Bits perhaps is the right name for such a type, but we'd have to refactor the class hierarchy, renaming current-Bits to GroundType or something, and making Bits, UInt, and SInt its descendants.

aswaterman, 2016-04-09:
(but, I think we are all in agreement that the current incarnation of the *type* Bits, if not the constructor, does not make very much sense.)

colinschmidt, 2016-04-09:
I'm not sure I understand what "true" additional saftey is? Is this based on most uses of Bits being replaced by Vec(Bool()) to add saftey?

sdtwigg, 2016-04-09:
The underlying issue is that bitwise operations between two things have mixed/'unknown' signedness resulted in too many surprising behaviors. Paradoxically, arithmetic between Bits is much more straightforward to define without surprising behaviors.
For arithmetic, the infectiousness of signedness isn't too surprising. If you add (assume bit-extending addition) a bunch of mixed signed numbers together then having a signed result is fine because you expect the sum may be negative. Also, those additions are all still commutative and associative, as expected, so you can rearrange into an adder tree with no issue.
For logical operations, the 'signedness' just decides whether you zero-extend or sign-extend and neither really trumps the other behaviorally: If you 'and' together a zero-extended number or sign-extended number, should the result later be zero-extended or sign-extended? No matter the decision, the operations are not associative. So, if you 'and' a bunch of (mixed sign and width) numbers together, you can't safely rearrange into a tree.
The problem with logical operations could be solved by prepending a 0 to the UInt before operating on it; however, this leads to a host of other problems and seems weird.
(Consider (UINT ^ -1).andR or when you and Bits of all the same, known width)
In general, for mixed signed logical operations, it is much safer to have the user call for the precise behavior they want. It should be rare enough that this isn't much of an issue. (And, a pad operation helps handle particularly weird cases where you want to mask a uint with an effectively sign-extended literal.)

As an example, recall the back-and-forth on what the Mux between UInt and SInt should return:
https://github.com/ucb-bar/chisel/commit/5418aa5ffedff213ffb7898524c722fb10fbddd5 (UInt)
https://github.com/ucb-bar/chisel/commit/53b7562ae776668f679830c05625ba48c38fdfa9 (SInt, current chisel2)
The problem usually arose for an N->1 mux where n > 2, particularly if one of the elements was -1 and others were UInts. Let's look at muxing between -1, 1, 3, 5. Assume this associativity (((-1, 1), 3), 5)
For the UInt behavior, the -1 will present as 0x001, not 0x111 as likely expected.
For the SInt behavior, the 1 will present as 0x111, not 0x001 as likely expected.

What about calling the ancestor Digital (in contrast to the 'soon-to-be-introduced' Analog types)?
you can then add the following to the chisel package object:
type Bits = UInt // alias UInt type
val Bits = UInt // alias UInt object

aswaterman, 2016-04-09:
@colinschmidt I meant using Bits vs. using UInt everywhere

colinschmidt, 2016-04-09:
Just to make sure I understand:

If we define Bits as always zero extended, we still run into issues when someone 'and's this mask signal with an SInt (value should be sign extended)? Among other similar cases.

aswaterman, 2016-04-09:
Defining Bits to zero-extend is part of the problem; that it is the superclass of SInt is then a contradiction.  If it were just another type lateral to UInt and SInt, then we could start to define how these operators behave.  (But people would justifiably have problems with allowing Bits to intermix with UInt and SInt; there would be a lot of explicit casts.)

sdtwigg, 2016-04-09:
I don't understand the question. Assuming you mean Bits as 'the ancestor of SInt,' you simply can't define it as always zero-extended because SInt, the subclass, will sign-extend. So, you violate Liskov.

aswaterman, 2016-04-09:
yeah, that

colinschmidt, 2016-04-09:
Ok I was thinking not of the current Bits but a new type more like what Andrew was saying. Something like a descendant of UInt that doesn't support arithmetic.
But I think I understand the problem now. Thanks everyone.

closed 2016-04-11 by colinschmidt