chipsalliance/chisel #136: Forcing width of data

colinschmidt, 2016-04-11:
Porting from Chisel2 I was able to force the width of a Node with something like
```
UInt(data, width)
```
This apply method has since disappeared and I ended up using something like
```
val data_wire = Wire(UInt(width=width))
data_wire = data
```
Is there a canonical way to do this? A function I should be using?

aswaterman, 2016-04-11:
Wire(UInt(width=x), init=value) should work in both.

Is there a FIRRTL node that returns a copy of x with the same width as y
(where both x and y have unknown width)?  That is probably more generally
useful.

jackkoenig, 2016-04-11:
Perhaps I'm not understanding the problem, but with the Chisel code:
```scala
class Test extends BasicTester {
  val foo = UInt(5, 32)
  val bar = UInt(23, 32)
  val r = Reg(init = foo)
  r := bar
}
```
I get the firrtl:
```
circuit Test : 
  module Test : 
    input clk : Clock
    input reset : UInt<1>
    output io : {}
    
    io is invalid
    reg r : UInt<32>, clk with : (reset => (reset, UInt<32>("h05")))
    r <= UInt<32>("h017")
```

Is this not what you're trying to do? Do you have a more specific example?

colinschmidt, 2016-04-11:
I've transformed them using the init parameter in Wire and it works fine.
Thanks Andrew.

closed 2016-04-11 by colinschmidt