chipsalliance/chisel #137: No default dynamicContext in Builder

chick, 2016-04-11:
This is not urgent, but as someone coming to Chisel from the scala side, I like to be able to go into the interpreter or scala worksheet and just fiddle around.  The default Builder dynamic context of none prevents this. Changing line 88 of Builder.scala from
```scala
  private val dynamicContextVar = new DynamicVariable[Option[DynamicContext]](None)
```
to
```scala
  private val dynamicContextVar = new DynamicVariable[Option[DynamicContext]](Some(new DynamicContext))
```
Let's me perform small experiments in interpreter, and seems to cause no harm to tests etc.  Could we consider adopting this change, or are there deeper considerations behind this design decision.

sdtwigg, 2016-08-31:
Just as a warning:

Using getOrElse instead of the proposed fix introduced an 'interesting bug' (related to idGen and the equals override). Any Data (and Modules, if you somehow avoid the .get failure exception) produced outside of elaborate will have id = 0. Thus, due to the equals override, they will all be == to each other (which, amongst other oddity, causes very erratic behavior when put in certain Scala collections).

chick, 2017-02-02:
Not worth keeping this open

closed 2017-02-02 by chick