chipsalliance/chisel #139: Support zero width wires

colinschmidt, 2016-04-12:
Hwacha makes use of zero width wires and I would prefer not to add lots of workarounds for a feature that will be eventually added.

Is there an ETA for this support?

jackkoenig, 2016-04-12:
The real issue is ucb-bar/firrtl#65

There is no current ETA but I am willing to work on it. It is somewhat difficult because this will require first laying out the exact definition under all operations, then adding support all over the place. 

Any help with defining them would be greatly appreciated in the linked FIRRTL issue

colinschmidt, 2016-04-12:
Ok. I'll put some comments there. I'm going to try working around them for a bit and see how many I encounter.

closed 2016-04-12 by colinschmidt