chipsalliance/chisel #141: Regs are allowed in io bundles. Does this make sense?

dt27182, 2016-04-14:
There is no programmatic mechanism to prevent Regs from being instantiated in io Bundles. I'm not sure what's supposed to be synthesized in this case.  Are there any legitimate use cases for this?

sdtwigg, 2016-04-14:
First, I don't think this makes sense in general and, due to a plethora of corner case issues in chisel3 and firrtl, think it should not be allowed at all. (Note, see my PS)

That said, you could feasibly infer a meaning for IOs if you really wanted and thus fabricate some use cases. It could mean that you have a register either right after the actual module input wire or right before the module output wire. (Things start getting messy when you flip though.)

PS: I am strictly thinking of the case where the register is created directly in the Bundle constructor (and thus being inferred as a field of the Bundle). I see no issue in defining a function inside the Bundle that creates a register as part of its execution. The function will be called within a module and thus the owner of the register is unambiguous.

aswaterman, 2016-04-14:
Anything that results in the emission of a FIRRTL statement should be disallowed in a Bundle constructor (Reg, Wire, arithmetic, etc.).  It's not obvious how to force this syntactically (and as @sdtwigg points out, you don't want to disallow this behavior from within method calls on the Bundle), but it should be straightforward to add a dynamic check.

closed 2016-05-31 by ducky64