chipsalliance/chisel #142: Width specification on IO ports

dt27182, 2016-04-14:
Currently, IO ports can be defined without widths and width inference in FIRRTL will take care of it. In process of porting over the Chisel 2 classic scala tester, we've come across the issue that we need to know the widths of the IO ports before FIRRTL is run. Regardless of this use case, it does make some sense to force the width to be defined on all the IO ports because the IO ports are the interface to a module and you don't have a fully specified interface without the width. However, there are use cases where sub module IO port widths should be inferred based on the context in which they are used. We should discuss this issue.

sdtwigg, 2016-04-14:
I'm of the opinion that all modules should have fully-defined port definitions.

For one, this allows you to more comfortably predict the performance of the width inference algorithm and constrain the scope of potential 'uninferrable width' issues.

colinschmidt, 2016-04-25:
I think that this should not be required. If a user would like to specify all widths that should be there prerogative but I don't see the benefit of forcing everyone to do this.

sdtwigg, 2016-04-25:
I'm amending my answer. I think in the long-term you would want to migrate to non-specified widths. I think of the current Module behavior like C++/go-functions, which require explicit input and output type signatures. Then, eventually you migrate to having the width inference algorithms be more intelligent and allow holes in the Module IO types. This brings the Module behavior to be more like scala/haskell type parameterized functions.

The reason I say long-term is that there are way more other issues in the way before it would make sense to spend too much time on this feature.

ducky64, 2017-02-06:
Solved. Explicit widths on IOs are no longer required and has been solved with deduplication.

closed 2017-02-06 by ducky64