chipsalliance/chisel #147: Source Information via Macros, RFC

sdtwigg, 2016-04-19:
We need scala source information carried into the FIRRTL representation of the graph (see note 1 for caveat). This essentially needs to be done with the macro system.

**Part 1: Basic Proposal with Implicits and Macro-based materializers**
The general idea is that chisel API functions will take an implicit SourceInfo parameter. For example, addition will become:
```scala
def +(that: UInt)(implicit sinfo: SourceInfo): UInt = create_operation(...)
```
SourceInfo.scala has:
```scala
case class SourceInfo(filename: String, line: String, column: String)
object SourceInfo {
  implicit def materialize: SourceInfo = macro Locators.injectSourceInfo
}
```
Note, the compiler ultimately sees:
```scala
a + b // before macro expansion and implicit resolution
a.+(b)(_root_.chisel.SourceInfo("file.scala","1","3") // after
```
The use of implicits for this is partly inspired by https://github.com/lihaoyi/sourcecode (I was using implicits for a loosely related EnclosingModule concept way earlier). I want to use a separate implementation, or at least Chisel-specific SourceInfo subclass of those classes, so there is less chance of conflicts with other libraries that may use that library.
I can later walk through an implementation of using macros to extract line/file information as shown in gama: https://github.com/sdtwigg/gama/blob/master/chiselfrontend/src/main/scala/frontend/implementation/macrodefs/TransformMacro.scala#L16 (ignore the EnclosingModule part, that is a separate thing).

The general idea of using macros plus implicits rather than just macros is twofold. First, API functions that otherwise don't need macro transformations don't need to use macro transformations. They can just get SourceInfo by taking an implicit. Second, libraries can 'hide' their definitions from Chisel source locators so that Chisel instead 'sees' the caller of the user library through standard implicit chaining. e.g.
```scala
def add_uints(uints: Seq[UInt])(implicit sinfo: SourceInfo): UInt = uints.foldLeft(0.U)(_+_)
```
This should just work; however, the + nodes in the generated graph will have source information from where add_uints was called, not from where is was defined. (The implicit parameter on the function is shadowing the default macro-based materialization of SourceInfo.) 

**Part 2: Extension using macro transformations to hide implicits**
Implicits do not play nice with chaining apply calls. Consider the following (somewhat silly but I don't want to define too many funcitons for this example):
```scala
myuint(5,0)(2,1)

class UInt { def apply(l: Int, r: Int)(implicit sinfo: SourceInfo): UInt = {...} }
```
You will get an error that 2, an Int, is not a SourceInfo. The Scala parser is interpreting the second set of parameters as an explicit override of the implicit parameters.

To resolve this case, we need to use the trick I used in gama. This is where I use macros, for example, to transform an apply or extract call into a do_extract call with extra arguments. The API similar to the following:
```scala
class UInt {
  def apply(l: Int, r: Int): UInt = macro XForm.to_do_extract
  def do_extract(l: Int, r: Int, sinfo: SourceInfo): UInt = create_operation(...)
}

myuint(5,0)(2,1) // before macro expansion
myuint.do_extract(5, 0, implicitly[SourceInfo])(2, 1, implicitly[SourceInfo]) // after macro expansion, before implicit resolution
```
In summary, I like to say the macro transform into do_* is making the implicits explicit and unifying the argument lists. I know this trick works because I tested it extensively in gama. However, in gama, every operation is transformed into a do_* form.

This may be unnecessary. You only need to transform operations that you expect to be occasionally succeeded by an apply (like apply). Some operations like addition are unlikely to. For example (a+b)(0) shouldn't be a problem due to the parenthesis surrounding a+b. Note that a.+(b)(0) is still a problem. Of course, for consistency, it may make sense to transform all operations.

Notice that in the Part 1 library case, you can't do add_uints(Seq(a, b))(0). To allow that, the library would have to use macros to transform into the do_* form as well.

**Notes**
Note 1: The proposals will, in general, only give *one* line of source information: the line that invoked the chisel API. If the user code involves libraries and intervening functions, this information may not be meaningful in all cases. For example, if I have a function that takes two parameters and just connects them together, then the source information will just point to that function definition. Thus, if FIRRTL reports an error, it will complain about that library code, which didn't necessarily make the mistake. Ideally, the frontend will still do sanity checks as much as possible since the frontend can dump the slightly more helpful stack traces.

Note 2: These proposals all use simple def macros and thus do not need macro paradise. This has a more comforting forward compatibility story.

aswaterman, 2016-04-19:
re: note 1: I think this is fine in practice, as the degree of non-Module nesting is usually fairly shallow.

re: note 2: Sticking to def macros seems prudent to me.

sdtwigg, 2016-04-19:
re: re: note 1: I am mainly targeting a few select functions in ChiselUtils.scala and UInt itself.
For example, you can implement UInt.andR with (this !== 0) and the implicit shadowing would ensure the proper source locations are captured. (In retrospect, I should have used that as the example rather than add_uints.)

ucbjrl, 2016-04-19:
I'd like to plea for some support (entirely optional) for at least a limited amount of stack tracing. When you encounter an error down in the bowels of a generator, it's extremely helpful to see the path that was used to arrive at the offending line.

sdtwigg, 2016-04-19:
You can and should still throw an exception at elaboration time. This is just to annotate the FIRRTL graph so that backend passes (post-elaboration) can have some source information.

The likely debugging flow when the backend complains is to first goto the offending lines. Then, they may either just see the necessary fix or add some extra checks/debugging information and then retry.

ducky64, 2016-04-19:
For Note 1, I think there's a good case for source locators for hardware generators in functions (ChiselUtil style). I think it's less a case of nesting and more a case of where you invoked a certain function, especially if it gets used a lot. I don't think frontend checks can capture all the use cases - it only handles cases where the Chisel code is bad, not where you're debugging why the Verilog isn't working or trying to figure out what synthesized to what.

One example would be if you instantiate several Counters in a module. Since they're hardware construction functions, they won't have clean Module boundaries, and just appear as logic within their encapsulating Module. It may be hard to tell which piece of logic belongs to which Counter,

One possibility might be to allow multiple source locators (this would also need a FIRRTL spec update), so that the source locator can have both a source line in the guts of the library and a source line where the library was invoked. I'm not sure if it's possible to build infrastructure that can automatically track where a library was invoked without slow reflection, so one solution might be to provide infrastructure and conventions so library writers can weave through source context information.

ucbjrl, 2016-04-20:
Right, but I think there's a tendency to do less error checking in Chisel3, under the assumption that FIRRTL needs to do it anyway. There appear to be only 6 calls to Builder.error() in Chisel3.

sdtwigg, 2016-04-20:
> Right, but I think there's a tendency to do less error checking in Chisel3, under the assumption that FIRRTL needs to do it anyway. There appear to be only 6 calls to Builder.error() in Chisel3.

So, this is a separate discussion that can easily overtake this issue (#150 might be a better place since it refers to a more direct consequence of the delayed error checking approach). Basically, I vehemently oppose the strategy of deferring errors. For error checking, redundancy is OK: Chisel3 can verify a property holds and FIRRTL can also independently verify it as needed.

A lot of the chisel literature/slides make reference to the design 'cycle' where chisel is supposed to give meaningful feedback to the user earlier. I consider this just a specific case of that: Give elaboration errors as early as possible while you still have access to appropriate context (like stack traces, etc.).

The macro proposal is just a means of carrying over non-zero amounts of context to later passes so that you can better handle errors which are too hard to catch mid-elaboration (like combinational loops).

sdtwigg, 2016-04-28:
Desire example of source info handoff to FIRRTL.

i.e. assume I have in UInt
def +(that: UInt)(/*stuff*/) = {
  val sfile, sline: String = /*stuff*/
  // what do I do with ifile and iline?
}

ducky64, 2016-04-28:
I think the FIRRTL spec has a section for source locators (page 51 of latest version, https://github.com/ucb-bar/firrtl/blob/master/spec/spec.pdf). It currently only allows for one source locator (it has explicit filename, line, and col sections) which might hamper libraries down the road, but that's probably fine for the first implementation here anyways.

I plan to add a Option[SourceLocator] (where SourceLocator is a case class) to each Def-whatever, and that will emit the source locator style per the spec.

sdtwigg, 2016-04-28:
Sounds good. I setup the sketch to meet you at that point.

ducky64, 2016-04-28:
So your example uses implicits for cases that are unlikely - what are everyone's thoughts on that? I was thinking to structure everything with explicit macros - it will be more lines of (boilerplate) code, but it will also mean more consistency, from both a implementation point of view and for anyone trying things that are a bit odd.

Then, for library writers, an API could be exposed (in a future feature) for them to append (or replace) source information from their library point of entry (though again, this needs a FIRRTL change). I think it's even less likely that libraries will be able to take implicit source information without issues with chained calls.

sdtwigg, 2016-04-28:
These are orthogonal concerns.

First, WRT FIRRTL changes, won't it just take a string as the info? Then the library can say the original source info is whatever it wants. Even now, the library can use any arbitrary string for the 'file'. To make this more obvious, I can add a second SourceLoc: 'RawString' that is just one string with custom-defined information. As a toy example, (which likely needs slight cleaning later), I say the line is "Queue" rather than a line number. (Note that getting the actual line number in a reliable manner will require separate compilation.)

Also, really, this is all just a hack around not being able to throw a delayed stack trace (by trying to reconstruct a weird partial one) and underlines the advantages of detecting issues and throwing an exception when possible. (My main fear from various conversations is that these SourceInfo items have been improperly built up as the panacea of Chisel3 debugging problems versus just being part of the remedy.)

Second, the use of implicits to actually materialize the SourceInfo is to allow library writers to use the basic non-transformed Chisel API (e.g. +, -, apply, Wire, etc.). Then, they can override the SourceInfo.materialize by shadowing it in the implicit resolution path. This is shown in the changes to Queue (where new information is substituted via implicit val) and Pipe (where the info is gotten from the user callsite into Pipe and then carried into Wire via implicit arguments). You want to keep this so that libraries can change the info while still using basic calls. (Otherwise you will have libraries written in one style of Chisel and components written in another.)

The transforms are to allow you to chain apply functions and only apply for classes that actually supply an apply function.
In the case of Pipe, it returns a ValidIO, which has no apply.
For Wire, it is true that T could have an apply; however, calling the apply immediately (ie. before binding to a val) has a very high chance of leading to an improper circuit anyway. For all the default-defined datatypes, it would return either a read extraction or write accessor (for Vec). In the former case, a write is never done (so unsourced wire) or, in the latter case, a read is never done (so the wire is never used anyway).

It is possible for the user to define their own datatype with an apply method that provides useful semantics when immediately called from the Wire return. Since there is a fear of that case, we can add a transform for Wire.

That said, it seems there are a lot more cases where the transform needs to happen than I initially projected. (The scala compiler will complain about (UInt(5)+UInt(5))(0), for example. So pretty much any operation returning an applyable type should be transformed). Even so, it just seems too heavy to transform every method in the base API and utils. (Particularly for utils, it would make them trickier to read).

ducky64, 2016-04-29:
The current FIRRTL spec has source locators spec'd as:
@["filename", line, col]
It's arguable whether everything in the `@[...]` is read as a string or if formatted data is expected. If formatted data is expected, then multiple locators aren't possible without some hacks (like putting everything in the filename)

My expectation is that multiple source locators will be used not for correctness debugging (which should indeed throw an error, "fail fast"), but for performance debugging ("what created this giant chunk of Verilog that's failing timing?"). That this helps for correctness debugging (until Chisel3 frontend is more bulletproof) is extra.

Did you have a way for transforms to play nice with implicits? The function that invokes the macro (probably?) can't take an implicit, since then the apply() chain will fail. Perhaps there's some way in the macro to resolve implicits in the enclosing scope without it being part of the function signature?

sdtwigg, 2016-04-29:
So, for the first point, just change FIRRTL, and this applies to a multitude of other potential issues. It is too early, there are too many issues/missing functionality to consider the spec locked in. If there is too much overhead in doing this (having to update the scala serializers/ANTLR deserializers), then make it a high priority issue to make doing iterative changes easier (switch to an easier serialization scheme like protobuf or just bypass the need to serialize altogether).

For the second point, you already have Module boundaries. Inside the Module are pointers to where in the library parts of it were instantiated and, at the Module instantiation site (and some of the glue logic), is where you can see where the user invoked the library. It may be a bit of an anti-pattern for a library to inject too much logic directly into the user's Module.

I don't fully follow the third point. Are you concerned about unrelated implicits? e.g.
```scala
def sum_uints[T](us: Seq[UInt])(implicit val blah: RandomUserType): UInt = us.foldLeft(UInt(0))(_+_)
```
This will still work fine. The implicit resolution will look for an implicit SourceInfo in the current imports (including the implicit val blah), see there is none, then move on to the next source (likely SourceInfo.materialize).

Chained applies is where you need to make the implicits more explicit (although you can still use implicit resolution via the implicitly trick):
```scala
myuint(5,0)(2,1) // before macro expansion
myuint.do_extract(5, 0, implicitly[SourceInfo]).do_extract(2, 1, implicitly[SourceInfo]) // after macro expansion, before implicit resolution
```

But not all operations need this because some return values that cannot be applied, like ValidIO[T] or DecoupledIO[T]. There is a related issue where some messiness in the chisel3 codebase adds a little extra work in applying and chaining extra calls; however, you can basically just soldier through it with the requisite boilerplate.

ducky64, 2016-04-29:
Ok, (I think?) I see now. Instead of having implicit as a parameter set, the macro adds `implicitly` which does the equivalent thing but inside the macro instead of at the interface (which prevents chaining).

sdtwigg, 2016-04-29:
Just to clean up the terms you used a little:

The macro transforms the old function call into a call to another (non-macro expanded) function that takes extra explicit arguments. The macro uses the old arguments as the appropriate arguments to the new function and add calls to implicitly to fulfill the extra explicit arguments.

Technically, neither the original or final function took implicit arguments.

ducky64, 2016-05-03:
Looks like macros still don't support named and default parameters:
https://issues.scala-lang.org/browse/SI-5920

This is a problem for a lot of Chisel functions, which require named arguments. I don't think there's a good way to both preserve chisel2 API compatibility and provide source locator macros for those constructs. Note that this also affects all uses of named arguments, not just where defaults are also needed.

One not-so-good option might be to just use implicits and hope no one chains an apply right after it, but that's probably also going to break compatibility.

sdtwigg, 2016-05-03:
I remember working around this by using boxes instead of named/default parameters.
e.g. https://github.com/sdtwigg/gama/blob/master/chiselfrontend/src/main/scala/frontend/Reg.scala (RInit, RInitModel etc. are the boxes)

The InitModel box might be droppable by just requiring the user to specify the init as the model as well for the cases where they really want it.

ducky64, 2016-05-03:
Yeah, I saw that in gama, but I think that would break a lot of the existing Chisel code, since it would both need people to instantiate boxes as well as drop the existing named arguments. Arguably, the boxing could make up for dropping names readability-wise, but if SI-5920 ever gets implemented, the recommended way would probably go back to named arguments.

sdtwigg, 2016-05-03:
(This is getting tangential, but:)
Overloaded method names with boxing is still preferable in the Reg case because Reg is so open-ended that there are no sensible defaults (all parameters have a default but you can't call the method with no parameters. Also, the current use of null is a style violation (and, in particular, does not work properly with type inference) and replacement with Option is just another form of boxing.

Amusingly, that Reg implementation was first written to address the type inference problem and resolving the macro issue was a collateral benefit.

ducky64, 2016-05-03:
For Reg, it seems that in ideal macro wonderland, the Reg apply call could take a default null and transform it to a Option[] type before calling do_apply (or something). Presumably, a macro could also encode rules for how to handle partial parameters in a way that can cause a compile time error if no parameters are passed to it.

ducky64, 2016-05-05:
For now, we're just going to not add source locators for functions that are commonly invoked with default or named arguments. This will include at least `Reg`s and possibly also data type constructors (where you might get `UInt(width=5)`).

In the long term, if named/default arguments doesn't happen anytime soon, we will consider introducing a new (hopefully also cleaner) API that supports source locators.

A related but different issue is proper naming support for nodes that aren't direct Module members, like when you declare wires in a scope. This can be a stopgap solution to provide better source information until named/default arguments gets resolved. A possible implementation may be with defmodule / defbundle style macros.

closed 2016-05-31 by ducky64