chipsalliance/chisel #148: .asInput not respected in Vec creation

colinschmidt, 2016-04-19:
```
...
      val wpred = Vec(2, new BankPredEntry().asInput)
      val wdata = new BankDataPredEntry().asInput
...
```
produces the firrtl

```
... wpred : { pred : UInt<8>}[2], flip wdata : { data : UInt<128>, pred : UInt<8>} ...
```

Where I would expect wpred to also have flip

sdtwigg, 2016-04-19:
I consider this philosophically related to the issue with DeqIO outlined in https://github.com/ucb-bar/chisel3/issues/104 where, essentially, writing clone methods (here the flavor is called cloneType) is hard in the presence of mutable state.

The advised way of writing a cloneType method in both chisel2 and chisel3 has usually been to just create another instance using the same constructor parameters. However, this will fail to capture changes to mutable state done after construction and thus not captured in those parameters.

For gama, I addressed this problem by providing a 'SimpleCopy' trait (copy:gama::cloneType:chisel3, roughly). So, user defined types only had to provide a 'structurally similar' copy returned by simplecopy, defined as all Vecs have the same number and type of elements, and Bundles had the same name and type of fields. Then, SimpleCopy would provide a copy function that:
1. Called simplecopy to get the 'structurally similar' copy
2. Walked the copy and original in tandem and
3. At each root (Element e.g. UInt, SInt, Bool, ...) ensured they had the same direction and width
(Width was because, why not, already there soooo.)

An error was thrown if the walk failed (saying the copy was not 'structurally similar').
Some sample snippets are here:
https://github.com/sdtwigg/gama/blob/master/chiselfrontend/src/main/scala/frontend/SimpleCopy.scala
https://github.com/sdtwigg/gama/blob/1d314e4cec7f804861a0c2cd7661c3eb298a145c/chiselfrontend/src/main/scala/frontend/Data.scala#L17
with e.g.
https://github.com/sdtwigg/gama/blob/1d314e4cec7f804861a0c2cd7661c3eb298a145c/chiselfrontend/src/main/scala/frontend/implementation/ElementImpl.scala#L16
https://github.com/sdtwigg/gama/blob/1d314e4cec7f804861a0c2cd7661c3eb298a145c/chiselfrontend/src/main/scala/frontend/implementation/HardwareTupleImpl.scala#L20
Tthe asSPEC parameter means return as 'unbound' or 'bound.' cloneType would in general want to return unbound so it could then be attached to a register, wire, etc. However, Vec wants a true copy so that created accessors for Reg(Vec(Bool)) look like a Reg(Bool).)

Technically, this shouldn't increase the algorithmic complexity of cloneType since new is already O(n) where n is nodes in the data structure. Also, you can dispense with the simplecopy/copy distinction. (I kept it because I hadn't yet settled on whether I wanted to allow the user to override with a 'better' implementation of the tandem walk) Finally, could clean up the error message strategy to augment the exception with information on what parts were detected to be different.

PS: The glory with simple copy was that I could have an '@bundle' macro annotation that automatically created a simplecopy method by reinvoking the constructor with the same parameters:
https://github.com/sdtwigg/gama/blob/master/chiselfrontend/src/main/scala/frontend/implementation/macrodefs/AnnoImpl.scala#L84
SimpleCopy took care of most of the 'edge' cases. (Of course, a malicious user could easily construct an annoying edge case. Like including a var bad: Data in the Bundle and then mutating it. However, then you get the structural differences exception!)

sdtwigg, 2016-04-20:
Adding @aswaterman since he was involved in the related issue.

ducky64, 2017-02-06:
Superseded by https://github.com/ucb-bar/chisel3/issues/440
asInput should clear Bundle directionality and make it all inputs.

closed 2017-02-06 by ducky64