chipsalliance/chisel #150: Bad Error Message when Wire, Reg not used

sdtwigg, 2016-04-20:
**Problem**
```scala
val internal_wire = UInt()
internal_wire := io.in * 2
io.out := internal_wire
```

You will get a FIRRTL error saying that internal_wire is not defined. The error is far worse to manage when the name wasn't detectable. This issue is particularly relevant when porting from chisel2 to chisel3; however, it could easily come up when writing new RTL.

**Proposal**
My advised strategy is for the Wire, Reg, Literal, etc. constructions to also tag the datatype's nodes with information of what they are in the hardware graph (wires, regs, literals, etc.). Then, operations like connects or arithmetic, throw an error when involving operands not actually on the hardware graph.

You run into an issue with ports (as all current code does val io = // raw Data instantiation here). This is where my proposal to declare IOs as part of the Module constructor arguments comes into play. (The Module constructor can then tag the granted IO with the port information).

aswaterman, 2016-04-20:
I would support a better (earlier) error message in this case but don't have time to implement it at the moment.  I'd happily review a PR.

sdtwigg, 2016-04-20:
What do you think about the Port/io (sub)issue I brought up?

aswaterman, 2016-04-21:
Is the I/O thing not just a special case of declaring-hardware-inside-bundles?

sdtwigg, 2016-04-21:
I'm thinking of this:
```scala
class TestModule extends Module {
  val io = new Bundle {
    val in = Bool().asInput
    val out = Bool().asOutput
  }
  val internalwire = Wire(Bool())
  val internalreg = Reg(Bool())
  internalreg := internalwire // case1
  internalwire := UInt(1) // case 2
  io.out := io.in // case 3
}
```

Case 1 is fine. Both internalwire and internalreg were created after invocations of Wire and Reg were able to decorate the input Bool with information that they are wires and regs, respectively.
Case 2 is similarly fine because UInt(1) is a literal and thus actual 'hardware'.
Case 3 is problematic. We don't know that io.out and io.in are actual hardware since, at the time they execute, at best we know they are just Bundle. (Namely, Bundle of what).

The 'hack' would be to record this as a 'potential problem,' wait until the Module is fully constructed (Module.pop or whatever), go in decorate io with details it is a Port (at the same time, we check to make sure everything is nicely directioned, etc.). Then, review all the potential problems again to see if they are actual problems. First, this is hacky and, second, every potential problem would ideally capture a stack trace that can be thrown in the case it is a 'true problem.'

My proposal was to do instead:
class TestModule extends Module {
  val io = Port(new Bundle {
    val in = Bool().asInput
    val out = Bool().asOutput
  })
  io.out := io.in
}
// or
class TestModule extends Module(new Bundle {
  val in = Bool().asInput
  val out = Bool().asOutput
}) {
  io.out := io.in
}
```

I favor the latter because it lets Module be type parameterized on it's IO. So, instead of Module we have Module[IOT]. e.g. Queue[UInt] <: Module[QueueIO[UInt]]. This simplifies writing libraries that want to wrap modules and still expose their IO to the type system. One use case is that you can build a generic test harness 'for free' (without having to introduce your own shim class that is constraining the IO for you). e.g. "QueueTester(dut: () => Module[QueueIO[UInt]]) extends BasicTester"

The other use is (and the one I am really targeting) is being able to seamlessly replace a module with a mocked version of it. Rather than relying on chisel to figure out that two modules are identical (which requires fully elaborating each version), just explicitly elaborate only 1 copy. Then, all other use sites use the mocked version of it.

ducky64, 2017-02-06:
Bindings system fixed this... a long time ago...

closed 2017-02-06 by ducky64