chipsalliance/chisel #151: Why does Top.cpp run 10 extra cycles?

sdtwigg, 2016-04-20:
This seems somewhat arbitrary. (At best, I can think of needing 1 or 2 to get waves after an assert fails.)

The main impact of this change is that it is harder to write a 'tightly timed' unit test.

jackkoenig, 2016-04-20:
It is completely arbitrary, and you're right, it was intended to see a couple of waves after an assert fails. 

We could make some "cyclesAfterDone" option that the cpp wrapper accepts as a command line option that is exposed in the Chisel3 testing environment.

aswaterman, 2016-04-20:
I'd advise against adding command-line options, but would support paring it to the minimum necessary to get a usable waveform.  (Is it 1 or is it 2?)

sdtwigg, 2016-04-20:
My current setup makes it a little tedious to get waveforms from chisel3 so I can't offer too much advice on that atm. Here is a copy-pastable module to play with:
```scala
class FailsFast extends BasicTester {
  val counter = Reg(init=UInt(0, width=4))
  counter := counter + UInt(1)
  val fail = Wire(Bool())
  fail := counter < UInt(4)
  assert(!fail, "failure!") 
}
```

Then, just fiddle with the extra time until the waveform viewer shows fail == 1 but without any extra cycles. (In theory, it seems like 1 should be sufficient: otherwise tempted to look into other fixes. Start with 0 extra time to see if we get 'lucky.')

1-2 extra cycles would be palatable since the test timeouts needed a little bit of wiggle room anyway to account for startup time. 10 cycles just seemed too large and arbitrary enough to be subject to change. Also, 10 cycles later might be too much later that the failure condition dropped. Actually, it is tempting to also request another signal be added (per assert) that goes high (or low) when the assert fails.

closed 2017-02-06 by ducky64