chipsalliance/chisel #152: Allow for bulk connect on Seq?

colinschmidt, 2016-04-21:
In chisel2 I was able to create a Vec of nodes and bulk connect them to another Vec of the same type as follows

```
  val issue = Vec(io.abox(0), io.pbox(0),
    agent.io.span(io.abox(1), io.pbox(1)), io.abox(2))
  issue <> agent.io.issue
```

In order to use chisel3 this is transformed to

```
  val issue = Seq(io.abox(0), io.pbox(0),
    agent.io.span(io.abox(1), io.pbox(1)), io.abox(2))
  issue zip agent.io.issue map {case(s,d) => s := d}
```

The issue Vec/Seq in this case will not be dynamically indexed itself but the agent.io.issue Vec will be, so it seemed reasonable to match the types for the previous <> usage.

Is this transformation the only way chisel3 will support this operation?

aswaterman, 2016-04-21:
We'd have to add an implicit conversion from Seq to a new type that has :=
and performs the same map.  But it is possible.

sdtwigg, 2016-04-21:
You can also do:
```scala
for((i, a) <- issue zip agent.io.issue) { i <> a }
```

ducky64, 2017-02-06:
Resolution: we should do this.

ducky64, 2017-11-08:
Does this work yet?

colinschmidt, 2017-11-08:
Why would it work know without anyone submitting a PR against this issue?

jackkoenig, 2017-11-08:
Well, this issue exists: https://github.com/freechipsproject/chisel3/issues/482

jackkoenig, 2017-11-08:
It looks like the issue is connecting to Seq as the lhs value which does require an implicit conversion as Andrew proposed

ducky64, 2019-11-05:
Discussed at the dev meeting today:
- apparently there is already support for Seq as the RHS of a connect
- so the work left is supporting Seq as the LHS of a connect
- some kind of implicit conversion would be required - whether a fully general Seq -> Vec conversion (and analysis of potential side effects) or a limited Seq -> LhsConnectable (or something similar that only provides the limited connect APIs) would be a point for discussion
- potentially a good starter issue, though limitations of implicits might make things interesting

jackkoenig, 2022-03-22:
Done in 3.5.0 via `DataView` and experimental import:
```scala
import chisel3._
import chisel3.experimental.conversions._

class SeqExample extends RawModule {
  val a, b = IO(Input(UInt(8.W)))
  val x, y = IO(Output(UInt(8.W)))
  Seq(x, y) := Seq(a, b)
}

println(getVerilogString(new SeqExample))
```
https://scastie.scala-lang.org/17fTL4WKTYOvPioFO6VOGA

closed 2022-03-22 by jackkoenig