chipsalliance/chisel #153: addPin from chisel2

sdtwigg, 2016-04-21:
chisel2.Module had an addPin method that allowed the user to add a pin to the Module outside of its io (Actually, it would add it to the io Bundle; however, I am proposing that it adds the pin in the same hierarchy as clk, reset, and io. Into chisel3.Module.ports).

This was useful (essentially requisite) for writing libraries that needed to thread signals across Module boundaries. As an analogy (to justify the behehavior of bypassing io), if io is similar to a function's signature, addPin is part of the architecture for throwing exceptions from inner Modules.

Ideally, the chisel3 addPin would actually check to make sure no accidentally overwrites occur.

jackkoenig, 2016-04-21:
Should this be part of Chisel3 or should things like this be handled as FIRRTL transformations?

ducky64, 2016-04-21:
I always thought this would be handled as a FIRRTL transformation in Chisel3. Chisel3 frontend no longer has the entire hardware graph like Chisel2, so it's harder to do anything but the simplest operations. The _parent links are also private, so if a goal was to propagate signals up the Module hierarchy, that would mean either exposing more implementation details or having a purpose-built API, neither of which sound like clean ideas.

sdtwigg, 2016-04-21:
Philosophically, I consider addPin as part of the exposed chisel2 API because it is not really used internally (the FAME Backend calls it once) and was actually documented. Removing it is a major backwards incompatibility not mentioned in any of the migration literature. (I say major because making up for the lost behavior requires, at least, a new, not-yet-written FIRRTL pass.)

Specifically adding addPin as a FIRRTL pass would be too odd. (You would essentially be adding a FIRRTL pass that takes a description of the pin and the module it should be in and then adds it into the module. This is something the frontend is beyond capable of doing given it already has the Module.ports infrastructure.)

So, it would make more sense to add a pass to implement the broader behavior desired (threading information across Module boundaries). Basically, I am concerned with the cleanliness of adding a FIRRTL pass that changes the functionality of the circuit. First, there is not yet any way to attach extra information to the Module or its nodes. Second, the FIRRTL passes all sit in a separate binary so passing this extra information would require writing serializers/deserializers for extra data and modifying the FIRRTL binary main functions to be able to intake the extra data (as well as modifying build flows to transit the data). Then, this sort of modification to the graph has to be done very early (particularly, before width inference) so I need a way to make sure future restructuring of FIRRTL passes keeps this pass in the right place. Finally, it seems like FIRRTL is in a lot of flux (the past month rewrote the tree mapping API) so maintaining this pass would require a bit of effort.

I can understand wanting to move this into being a FIRRTL pass (and can see broadly how you would do is.) However, doing that seems like it would require a lot more work and I was hoping to restore this behavior sooner rather than later (even if future work transitioned it to being a FIRRTL pass).

sdtwigg, 2016-04-21:
The one thing that concerns me about a FIRRTL pass is that this would do functional changes to the circuit. This partly inspired my insistence that the pass has to run first.

As far as I know, none of the other passes actually do this yet. Also, this would blur the line on whether the 'extra information' is actually 'extra' since it is necessary information to describe graph function. Similarly, I am hesitant to call the extra information a type of annotation for the same reason: annotations thus far haven't been necessary to describe the function of the circuit.

ducky64, 2016-04-21:
Yeah, I agree that addPin as a FIRRTL pass doesn't make sense, but rather you want the higher level goal (like "hook up BIST pins") as the pass.

Anyways, do you still have enough non-private information in Chisel3 frontend to make proper use of addPin? Is this actually an objectively good API that should continue to be supported, especially considering how hardware graph is now private in Chisel3, that FIRRTL will have annotations (someday), and that FIRRTL and Chisel can actually be in the same binary (just need to change the Chisel3 emitter to directly generate FIRRTL data structures).

Assuming reimplementing addPin is the least-worst option if you needed this functionality _now_, what would be the plan for getting rid of it once FIRRTL transformations are straightened out? You probably aren't the only user of addPin, and others uses may not be clean or principled. Continuing support for it will mean more inertia and a less clean API break from Chisel2 to Chisel3, and we may end up having this discussion again when we do want to nix Chisel3 addPin in favor of FIRRTL passes for good.

sdtwigg, 2016-04-22:
I am still very concerned over introducing passes that are changes to the functionality of the graph. Given that firrtl does not yet support custom passes (and even then, this one is particularly sensitive to position), I would have to change firrtl anyway. But then, if I have to fork something, it would end up being far less work in the long run to change chisel3.

It isn't clear where the line is regarding hardware graph elaborations. I agree it makes sense to use a FIRRTL pass for transformations and adjustments that require a holistic view of the hardware graph. However, in this case, the adjustments can be safely done on a partially-elaborated graph.

Also, there is still the matter of timeline. Firrtl does not seem to be in a state to reliably support this pass at this time and it is unclear how long it will be until it is given all the pre-requisites. I'm unable to table this until firrtl is sufficiently fixed up.

ducky64, 2016-04-22:
So is your objection to functionality-altering passes on principle or because you don't think the implementation isn't ready yet (or both)? If the first, aren't passes just a more structured way of dealing with the circuit graph that was exposed in Chisel2? Can passes be structured in a way that limits scope (say, only altering a Module's contents) for additional safety? 

Anyways, I think there are several separate questions that we need to come to a consensus here:
- Is FIRRTL the best solution for doing this, in the long term?
- Is addPin actually an objectively good Chisel3 frontend API that we should continue to support?
- If addPin won't be part of the Chisel3 frontend API, what is the short term solution and what is the transition plan away from it (accounting for that people don't always want to fix their code to work with a new version, and that the Chisel2 -> 3 transition may be the best time to make breaking changes)?

sdtwigg, 2016-04-22:
On the initial question, it is moreso the latter. If I was forced to assume FIRRTL was etched in stone, the former would definitely apply. Although, there definitely are philosophical concerns with trying to encode custom, mandatory functional information into the graph.

Part of the confusion is that chisel3 draws a very firm line between 'chisel3' and 'firrtl' as emphasized by them living in separate binaries and repositories. It seems to make the most sense to pass a fully specified design across that boundary.

1. I contend it isn't the 'best.' Where do you draw the line on what is a chisel3 feature? Chisel3 is currently weak on multi-clock domains. Should it just drop all multi-clock domain support and tell the user they need to use a FIRRTL pass to rewire the clocks in their circuit?

Further, it is a matter of using the proper tool for the job. Even if FIRRTL was more polished, the pass solution is 'heavy:' it uses a completely new API and adds a the backend component separate from the frontend components. (I also don't want to understate the increased risk in modifying the build flow in two places and ensuring the proper information is transited between them.)

For example, a scala library should prefer exposing an API of regular function calls to internal components versus a macro annotation that rewrites the user source code to invoke their internals directly. Similarly, chisel3 uses scala reflection in places rather than investing in a compiler plugin that dumps the raw typer output.

2. My main critique is that addPin (and related operations) is still part of building/appending to the hardware graph. You are just adding more information versus not transforming/changing already present nodes. I don't need a holistic view of the completed graph so it seems silly to wait.

While I acknowledge there are some advantages to this 'feature,' the chisel3 frontend already does allow one to transform the hardware graph via unguarded re-assignment to wires and registers. The old assignment is then completely lost (although, oddly, it may still interfere with width inference).

3. I don't seem a stable short-term solution that involves a FIRRTL pass. Also, is the chisel3 API already considered 'locked'? It looks like there is still plenty of cleanup to do on other parts, particularly for earlier error detection and reporting.

ducky64, 2016-04-22:
I don't think the Chisel3 API is considered "locked" until we do a public non-alpha (or even non-beta) release. And even then, I imagine it will evolve like any language, but probably with more emphasis on not breaking backwards compatibility.

Another question could be what exactly a fully specified design means. For example, a general pass could be developed that makes cross-module connections across any pins based on annotations. So in a sense, the Chisel3 frontend emitted design (with annotations) is fully specified, the FIRRTL just lowers the annotations to wires.

I'm not sure there is a very clear line between what Chisel3 and FIRRTL does, at least in terms of user-facing functionality (in terms of implementation, I think it's building the hardware graph one module at a time vs. traversing the graph with a complete view of all modules - obviously, some things straddle the line). Perhaps this is something that should also be better documented (much like the Scala-land vs. Chisel-land split)?

As for addPin, I think it's arguably "unsafe", especially in the context of a statically-typed language. Normally, you expect the io to be very well defined per-class, and you don't expect this to change across instances. addPin breaks those guarantees, so now the clean Module-as-a-blackbox-with-specified-io abstraction might not hold anymore, since you can muck around in a Module's externals from outside.

Are there other ways to accomplish the same thing without addPin? Like actually threading the necessary pins through explicit io? Or do you really need runtime-dynamic pins?

sdtwigg, 2016-04-23:
The idea was not to add things to the io Bundle but instead to the ports of the Module.
Module by default has ports io, clk, and reset. This would add ports at that level.

It would all still be statically typed as the val storing the added pin would still retain the appropriate static type and the type of io would not be changed. chisel3 is clearly still building the entire Module graph and then serializing it into a FIRRTL data structure so it isn't clear what distinction you are making there.
The fact that the Module internals are being supplemented by external actors is intentional since the goal  is to avoid polluting the module body and io with this side channel information. For the same reason, using the explicit io would not work.

There may be ways to make the hiddenness partially explicit, similar to how the IO Monad works to expose IO side-effects to the type system; however, this would require even more extensive reworks of Chisel internals.

sdtwigg, 2016-04-29:
https://github.com/ucb-bar/chisel3/blob/09958f63470697188e1ed1a01c7ea39b8c56e7ef/src/main/scala/Chisel/Aggregate.scala#L307

ducky64, 2017-02-06:
Programatically generated cross-module connections should be implemented as a FIRRTL transform.

We need to make that a standard chisel3 feature with annotations.

azidar, 2017-02-22:
This is ostensibly fixed with a combination of Record, and/or the proposed Module Hierarchy change with multiple IO's so I'm going to close this as a duplicate issue.

closed 2017-02-22 by azidar