chipsalliance/chisel #160: Chisel 2 compatibility: assignment of UInt to Vec

dt27182, 2016-04-28:
I'm porting the LBNL OpenSoC code to Chisel and have come across a statement that looks like:

val a = Vec(UInt())
a := UInt(0)

Chisel 2 actually assigns UInt(0) to each element in a. Chisel 3 doesn't support this type of assignment. Should this be supported in Chisel 3? If not, we should change the chisel 2 compatibility mode checker to flag this as an error.

Personally, I don't think this type of assignment makes a whole lot of sense, but would love to hear everyone's thoughts on the issue.

aswaterman, 2016-04-28:
Seems a little hackish... I'd favor not supporting it in chisel3 and instead suggesting something more explicit like

    val a = Vec(UInt())
    a foreach { _ := UInt(0) }

aswaterman, 2016-04-28:
There's also

    val a = Wire(init = Vec.fill(n)(UInt(0)))

sdtwigg, 2016-04-28:
Definitely agree with Andrew's alternative over the former. The first is too much of a type hack.

dt27182, 2016-04-29:
The 2nd approach looks good to me as well.

closed 2016-05-31 by ducky64