chipsalliance/chisel #162: chisel2 vs. chisel3 <>, := semantics

sdtwigg, 2016-04-29:
I can get into examples if necessary: But, basically, <> and := work very different in chisel3 than in chisel2.

In chisel2, := is a mono-directional connect operator where you knew the LHS was a sink and the RHS was a source. <> is a bi-directional connect operator where is figured whether nets were sources or sinks. (This required at least one side to be an IO so the appropriate direction hints were available.) A nice property of <> was that it was commutative.

In chisel3, the semantics are a bit more topsy turvy. In code, it seems like <> is often just an alias for :=. I keep running into oddities and bugs related to the edge cases for this. (In old code one of the structs was defines as all being inputs instead of all being outputs and this led to really crazy verilog.) Also, in some cases, it isn't clear which argument has to be on the LHS for a <> connection between two module IOs.

Given that chisel3 tries to maintain compatibility with chisel2, what are the drawbacks of restoring the chisel2 behavior in chisel3? I'm willing to go through necessary measures to fix this.

aswaterman, 2016-04-29:
I believe the intent was to deprecate <> and only have a single connect operator, :=, which is bidirectional.

Regardless, <> is mis-named, because it isn't auto-sensing like the chisel2 one.

aswaterman, 2016-04-29:
Overall, I preferred the commutativity of <> in chisel2.  It would be nice if such an operator were the only connect operator in chisel3.

sdtwigg, 2016-04-29:
I see. Philosophically: why? There are a lot of weird behaviors in Chisel2 but := and <> were actually pretty reasonable*. It isn't totally clear how a single unified bidirectional operator would actually have easily explainable semantics. 

Practically, when? The current state of affairs is dire. The timeframe will reflect whether this is worth addressing.

aswaterman, 2016-04-29:
I'd say the current state of affairs is radiant.

aswaterman, 2016-04-29:
I dunno, I just think it would be less confusing if there were only one.  := and <> do basically the same thing as each other in chisel3, and we're getting along just fine.

aswaterman, 2016-04-29:
Deprecating one of them would have to wait until we've migrated off of chisel2 entirely, though, since it's basically impossible to write chisel2 code without using both.

ccelio, 2016-04-29:
Maybe I'm just an idiot, but `:=` being bi-directional is really kind of confusing. Even the operator itself isn't symmetric!   But I can sort of see if, under the covers, the two operators are one and the same then it's a harder sell for keeping both.

sdtwigg, 2016-04-29:
This came up in the following snippet:
```scala
class InnerModule extends Module {
  val io = new Bundle {
    val intake = new Bundle {
      val port = Decoupled(Bool()).flip()
    }
  }
  // logic
}

class OuterModule extends Module {
  val io = new Bundle {
    val intake = new Bundle {
      val port  = Decoupled(Bool()).flip()
    }
  }

  val i1 = Module(new InnerModule)
  i1.io.intake <> io.intake
  // more logic
}
```
The intent is that i1 is 'consuming' the part of the parent io.
However, I have to write io.intake <> i1.io.intake.
This inverts the consumer <> producer format I'm used to using for <> and :=.

dt27182, 2016-05-25:
I have the same issue with this in porting the OpenSoC to Chisel3. <> was used as bidirectional connect for many io bundles with both inputs and outputs in them and now the lhs/rhs order matters and causes the code to break in Chisel3. 

I think making <> directional for bundles, which is the current behavior in firrtl, does not make semantic sense as io bundles can contain both input and output members. Assigning a direction to the containing bundle is nonsense when the members are both input and output. 

Also from a practical usage perspective, as Stephen's example shows, figuring out the right order of <> connects with highly nested io bundles containing both inputs and outputs gets confusing fast.

sdtwigg, 2016-05-26:
Just to make sure things are properly linked:
This is addressed by #200

ccelio, 2016-07-29:
I wanted to poke this and see if there has been any movement towards a consensus on "<>" and ":=" semantics.  

Firstly, the lack of commutativity is frustrating and un-intuitive (https://github.com/ucb-bar/chisel3/issues/243), but manageable - since it's caught at compile time.

But I'm just now trying to get BOOM updated to chisel3 and the new bulk-connection semantics are driving me insane. I have quite a few situations where a wide I/O bundle has a high fan-out to many modules, but most of the modules either don't write to it, or write to a small subfield, and yet the last connector dominates everybody else. This gets further tangled up with deeply nested Bundles with DecoupledIOs (inputs and outputs going in both directions at the lowest level of the tree), which make it exceedingly difficult to pull out just the signals any particular module needs to write to.

I know Jim is slowly working through @sdtwigg 's #200 (as seen here: https://github.com/ucb-bar/chisel3/tree/sdtwigg_connectwrap_renamechisel3), but I wasn't sure if there was a consensus or any further plans for bulk-connects.

ccelio, 2016-08-03:
BOOM is now updated to Chisel3 and passing all of the riscv-tests and I thought I'd share the experience.

Out of ~80 of the bulk connect uses, there was only 2 that required modification. But my vote is still in support of the old Chisel2 bulk-connect semantics.

**Old setup:**
````scala
io.imem <> fetchunit.io.imem
io.imem <> bpd_stage.io.imem
````

`imem` is a modestly wide interface that contains Bundles and DecoupledIOs.

Of note is a DecoupledIO named `imem.resp` that both `fetchunit` and `bpd_stage` need to read, but only `fetchunit` drives the `ready` signal to back pressure the DecoupledIO. 

**New Chisel3 Version**, I attempted to break out `imem` into it's sub-fields and only connect what's needed, but it still falls apart because of the lowest sub-fields still have I/Os going in both directions.

````scala
io.imem <> fetchunit.io.imem

bpd_stage.io.imem_resp <> io.imem.resp
io.imem.resp.ready <> fetch_unit.io.imem.resp.ready // required thanks to last-connect semantics

bpd_stage.io.btb_resp <> io.imem.btb_resp
io.imem.ras_update <> bpd_stage.io.ras_update
bpd_stage.io.npc <> io.imem.npc

io.imem.ras_update <> bpd_stage.io.ras_update
````

aswaterman, 2016-08-03:
Glad it worked out.  That was also the only such instance in rocket (where
the signals came from both the control and datapath).  I ended up merging
the control and datapath into a single module, which substantially improved
the readability of the code.

I'm ambivalent as to which semantics are better, but FIRRTL doesn't
directly support the chisel2 semantics.  It might be possible to implement
this operator as an add-on library to chisel3 that recurses through
aggregate types and applies := based upon the directions of the matching
names.

mwachs5, 2016-08-22:
Based on this current status of this issue, what is the advice for newly written Chisel3 code? Avoid <> altogether?

ducky64, 2017-02-06:
Current (chisel3) behavior needs to be documented in Data scaladocs.

We will leave behavior as-is.

jackkoenig, 2017-04-18:
I believe that this issue was made obsolete by stricter `<>` in `chisel3._`, however, there are remaining issues that are discussed in other issues:
* #495 
* #582

I plan to close this after writing a ScalaDoc for `Data.<>`

jackkoenig, 2017-06-26:
Also check that we have good testing specification for what `:=` vs `<>` mean

ducky64, 2017-11-08:
We've resolved this, additional proposed changes to connect semantics should be in their own issue.

closed 2017-11-08 by ducky64