chipsalliance/chisel #164: Separation of Utils and Frontend

ducky64, 2016-05-03:
With macro-based source locators happening ( #147 ), there now needs to be a clean separation of the Chisel frontend (core data types - Bool, UInt, etc) and things which invoke the frontend (like utils - Mux, Cat, PopCount, and friends) because macros require definition and invocation to be in different compilation units.

I think it makes sense for Chisel utils be strictly dependent on chisel frontend since it provides an example of how libraries can be written and structured (rather than invoking Chisel frontend internal functions directly, bypassing the source locator macro layer). If necessary, utils can also have transform macros for source locators, and may demonstrate multiple levels of source locators.

Anyways, the current problematic calls I've found are:
- `Data.toBits` is dependent on `Cat`
- `Vec.count` is dependent on `PopCount`
- `Vec.{indexWhere, lastIndexWhere}` are dependent on `PriorityMux`
- `Vec.onlyIndexWhere` is dependent on `Mux1H`

This issue was brought up about a year ago, during the Chisel3 code review, but that was before separate compilation needed to happen, so the consensus then was to leave it as-is. That's not necessarily a good option anymore. Thoughts for solving this?

aswaterman, 2016-05-03:
@ducky64 I think it's fair to reverse some of these dependencies... let me look into it.

aswaterman, 2016-05-03:
Can Chisel.util depend on something that's private[Chisel]?  Or is there some way of making stuff visible to Chisel.util but not to users of Chisel?  I'm thinking the easiest thing to do is to move the implementations into an object that lives in Chisel, and have e.g. both Vec.count and PopCount call SeqUtils.count:

    private[Chisel] object SeqUtils {
      def toUInt[T <: Data](x: Seq[T]): UInt                   
      def count(x: Seq[Bool]): UInt 
      def prioirityMux[T <: Data](cond: Seq[Bool], value: Seq[T]): T  
      def oneHotMux[T <: Data](cond: Seq[Bool], value: Seq[T]): T     
    }

sdtwigg, 2016-05-03:
@aswaterman Yes, subpackages can view parent package private values.

aswaterman, 2016-05-03:
Ok I'll make a PR

aswaterman, 2016-05-04:
@ducky64 anything else that f9dd3feae5727424b85b18cd6064140403b0411b doesn't handle?

ducky64, 2016-05-05:
From #168, looks like a sufficient split. Since SeqUtils will be placed in Chisel frontend, it will also have to be written without the nice source locator macros (so instead of toBits, it would be do_toBits, and their type signatures would take an explicit source locator parameter)

commit f9dd3fea (2016-05-03): Remove dependences from Chisel core on Chisel utils

Partially resolves #164
commit cd951e19 (2016-05-04): Remove dependences from Chisel core on Chisel utils

Partially resolves #164
PR #166 (2016-05-04): Remove dependences from Chisel core on Chisel utils
Partially resolves #164
closed 2016-05-09 by ducky64