chipsalliance/chisel #169: Vec(List(...))(index) no longer truncates index to log2(List.size)

colinschmidt, 2016-05-05:
Example code
```
Vec(pipes.map(_.res.data))(wsrc)
```

Where pipes is 4 elements long and wsrc is 3 bits. Both Chisel2 and Chisel3 agree on the width of wsrc but they create this indexing differently

Chisel2 takes the low order bits

```
assign T28 = T29 ? ifpu_io_out_bits_data : fpmu_io_out_bits_data;
assign T29 = T30[0];
assign T30 = T358;
assign T358 = wsrc[1:0];
```

While chisel3 does comparisons with constants without subsetting the bits.

```
assign GEN_0 = GEN_92;
assign GEN_114 = 1'h1 == wsrc;
assign GEN_90 = GEN_114 ? T_909_1 : T_909_0;
assign GEN_115 = 2'h2 == wsrc;
assign GEN_91 = GEN_115 ? T_909_2 : GEN_90;
assign GEN_116 = 2'h3 == wsrc;
assign GEN_92 = GEN_116 ? T_909_3 : GEN_91;
```

I worked-around/solved this issue by fixing the width of wsrc to 2. I'm not sure if Chisel3 wants to support this old functionality or not.

colinschmidt, 2016-05-05:
I wanted to clarify why I think this is not a firrtl issue:

```
T_911[wsrc]
```
is the firrtl code to select an element from the Vec and with wsrc.

The firrtl spec claims:

> A connection from a subaccess expression can be modeled by conditionally connecting from every subelement in the vector, where the condition holds when the dynamic index is equal to the subelement’s static index.

So it would seem that it is explicitly saying it will do this. I would also argue that perhaps this case should be mentioned in the firrtl spec but that is probably another issue.

aswaterman, 2016-05-05:
I think I favor address modulo the size of the Vec/Mem (rounded up to the next power of 2).

Why not make modulo addressing the behavior of FIRRTL?  It's theoretically fixable in Chisel3, but that's a weird place to do it.

colinschmidt, 2016-05-05:
@azidar Why does FIRRTL not do modulo addressing?

I agree with you that fixing this in Chisel3 is weird and moves away from a simple frontend (vec access is just passed through now)

azidar, 2016-05-05:
Hmm. In my opinion, modulo addressing is super unintuitive, and what I imagine the common case is actually a mistake (you didn't make your Vec big enough), not an intentional use of modulo addressing. In addition, it is trivial for the user to do the modulo-addressing themselves. Also, Verilog does not do modulo addressing. Any other thoughts?

azidar, 2016-05-05:
Also, I want to write a static checker that warns me if my index is out of bounds - this is impossible if that is technically legal behavior.

aswaterman, 2016-05-06:
OK, that's fine, we'll just do it in Chisel.

Making a static checker for array bounds checking is probably impossible.

azidar, 2016-05-06:
You're right, I meant dynamic.

aswaterman, 2016-05-06:
Haha ok that makes more sense :)

ducky64, 2016-05-09:
Referencing #176 

Is implicit modulo addressing the best idea? It seems that silently discarding bits is asking for trouble, and would be something a compiler would typically warn on (but since we don't have warnings, this probably should be an error).

On the PR, you had a function truncateIndex - how about making that a Chisel util so users would call that explicitly whenever there is a width mismatch?

aswaterman, 2016-05-09:
It's what you're gonna get from synthesis, whether you like it or not.  Doing it in Chisel at least codifies the behavior is the same between synthesis and simulation.

ducky64, 2016-05-09:
Just because it's going to be done in synthesis / simulation doesn't mean we can't make Chisel3 safer (but slightly more verbose). This also leaves it up to the user, who can either take the quick but dangerous approach by spamming truncateIndex everywhere or whether to reason through everything and size things correctly.

aswaterman, 2016-05-09:
In order for Chisel3 to actually be made safer, the FIRRTL code generation would need to change.  As it is, we will generate code for Mem that does different things in simulation and synthesis.  That is distinctly worse than doing modulo addressing.

Also, modulo addressing is consistent with assignment from wider to narrower types performing truncation.

I think you can make a justifiable case for changing FIRRTL code generation for Mem, and you can make a justifiable case for doing modulo addressing, but the status quo is probably not OK.

aswaterman, 2016-05-11:
I have a new proposal, which doesn't really change the status quo, but might clarify how we think about it.

Vecs have relatively simple behavior.  Write addresses are fully checked, so that out-of-bounds writes have no effect.  Out-of-bounds reads are defined to return random values in simulation.  In synthesis, read addresses are truncated (i.e., modulo-power-of-2 addressing), with an undefined value returned for out-of-bounds reads to non-power-of-2 memories.  Write addresses are never truncated, so writes never corrupt memories.

Mems have more potentially undefined behavior.  All addresses are truncated (i.e. modulo-power-of-2 addressing).  Out-of-bounds writes for non-power-of-2 memories are undefined and might corrupt arbitrary state in the memory (the simulation model might enforce this by randomizing the whole memory).  Out-of-bounds reads for non-power-of-2 memories return random values in simulation and undefined values in synthesis.

ducky64, 2016-05-11:
Any reason to have different behaviors for Mems and Vecs? Inconsistency seems like it could be a source of confusion ("My RTL is broken; Vecs did that, why don't Mems do the same?!?!1111").

I'm still strongly in favor of not discarding bits and making Chisel3 / FIRRTL error if truncation is needed and requiring the user to explicitly write a truncate. I think the extra verbosity is a good trade-off since then the powers-of-two case can be statically proven safe. For the non-powers-of-two case, there are several options, from generating dynamic checks to prevent writes (safe, but possibly resource inefficient), generating simulation asserts (likely won't catch all buggy code unless tests are through and designers know explicitly to test for this case), or just leaving it undefined (unsafe, but efficient if you assume designers can write error-free code).

aswaterman, 2016-05-11:
Mems already have different behavior (write -> read for SeqMem; colliding writes for all Mems).  It's already a caveat-emptor situation.

ducky64, 2016-05-11:
It not being perfect doesn't mean we have to make it worse though.

Were there actually good reasons for having this behavior, as opposed to just reasons why it isn't that terrible of an idea?

aswaterman, 2016-05-11:
We had talked about opting into the more performant, less safe behavior, and I think this is a reasonable vehicle for it.  In reality, regular memory structures in standard implementation technologies have limitations, and Mem was meant to target them at a low level without compromising performance.  I don't think my proposal is inconsistent with that ethos.

ducky64, 2016-05-11:
Ok. What about explicit vs. implicit truncation then?

aswaterman, 2016-05-12:
As I've said repeatedly, I think implicit truncation is perfectly justifiable.  I think a static error in FIRRTL is also justifiable, but is inconsistent with silent truncation on connections.

ducky64, 2016-05-12:
FIRRTL has the option for both (strong) connect and partial connect, whereas I don't think there's a similar option here right?

aswaterman, 2016-05-12:
That's a good point.

colinschmidt, 2016-06-09:
I made a PR to rocket turning the only implicit truncation that causes me problems into an explicit truncation. 
Either merging ucb-bar/rocket#44 or ucb-bar/chisel3#176 would be nice from a chisel3+rocket-chip user's perspective.

aswaterman, 2016-06-09:
I merged the rocket PR to deal with this for the time being.

ducky64, 2017-02-06:
Chisel3 truncates now by #176, no checks happen.

closed 2017-02-06 by ducky64