chipsalliance/chisel #170: RegNext creates wrong firrtl type

colinschmidt, 2016-05-06:
This chisel 
```
val z = RegNext(next = UInt(0), init=UInt(0, width=32))
```
generates the following FIRRTL
```
reg z : UInt<1>, clk with : (reset => (reset, UInt<32>("h00")))
```

Note that z is only 1 bit even though its reset has a larger width.

In Chisel2 this generates a 32bit register.

aswaterman, 2016-05-11:
Yeah, looks like RegNext doesn't match Reg(next=).  Can you confirm that the fix-regnext branch fixes the problem?

colinschmidt, 2016-05-11:
Yes that branch fixes the problem I now get a 32bit reg in verilog. The initial firrtl has no width but is firrtl infers it correctly.

```
reg z : UInt, clk with : (reset => (reset, UInt<32>("h00")))
```

closed 2016-05-11 by colinschmidt