chipsalliance/chisel #185: Things in chiselFrontend don't make it into the Scaladoc

jackkoenig, 2016-05-11:
This means UInt, SInt, Vec, etc. don't make it

sdtwigg, 2016-05-11:
On a related note, it is a bit of an inversion that chisel/chiselFrontend/src... has all the main Chisel._ stuff
and chisel/src/... has periphery stuff (like testers, utils, and Driver).

If utils was split out, would you even need to segregate for macros purposes? (since Driver doesn't need to invoke macro implementations)

ducky64, 2016-05-12:
See the source locator PR notes ( #186 ) for arguments of why splitting is nice.
In short, the only thing that *must* be split out are the transform function definitions. The macro use sites (`def blah = macro lolcats`) apparently don't need to be split out. However, I think there are good reasons to have the frontend split (notably, prevent `materialize` from being generated within the Chisel core, but allow it within utils).

sdtwigg, 2016-05-12:
I know it needs to be split. My argument was that the core (with all the main chisel stuff) should be in chisel/src and then the other stuff is in chisel/ChiselUtils/, etc.

ducky64, 2016-05-12:
I think that could make sense. So PR186 would be rewritten to be have:
(repo)/coreMacros/src/..., defining the transforms and nothing else
(repo)/src/..., main Chisel data types and `SourceInfo.materialize` definition
(repo)/chiselUtils/src/..., only the stuff currently in (repo)/src/.../utils
?

sdtwigg, 2016-05-12:
Yes, although, I would also put SourceInfo and SourceInfo.materialize in coreMacros.

ducky64, 2016-05-12:
The reason I didn't want to put `SourceInfo.materialize` in coreMacros is to prevent it from being implicitly invoked inside /src/ (frontend, definition of UInt, Bits, and friends), which is almost certainly a coding oversight (and having a source locator pointing you into the Chisel guts isn't helpful). This way, it will give a "can't use macros in the same compilation unit it was defined" error.

(Scala devs eventually plan on removing that restriction, but while it's there, we might as well make the most of it)

ducky64, 2016-05-12:
Anyways, I think it makes sense for #186 to get reviewed and merged first, then we can decide how to restructure things. Don't want to make an already-large pull request even larger.

ducky64, 2016-11-21:
This seems fixed now.

closed 2016-11-21 by ducky64