chipsalliance/chisel #188: Vec of size 0 not supported

ccelio, 2016-05-12:
I have been told that Vec's of size 0 is not supported.  I would like for them to be supported.

I have a pipeline of a parameterized size `num_stages`:

```scala
val r_valids = Reg(init = Vec.fill(num_stages) { Bool(false) }) 
`````

It may even be of size zero, in which the data passes through instantly with no registering.

ducky64, 2016-05-12:
How do you plan on using this? When it's zero size, you would have zero width wire (I think work is in progress to support that), but how would you get the combinational passthrough behavior? Presumably, if you tried to do anything with the Vec that's not a collective operation, it would just give you an error.

At a glance, I don't see anything that would prevent having a size zero Vec; the collective operations all seem to do approximately sane things on an empty Vec, and indexing into it hopefully should just error out.

Since you want to see both combinational and registered behavior, might it make more sense to define your own helper function / class, like makePipelineStage(numStages: Int) which internally can generate either a Wire or Reg(Vec)?

ccelio, 2016-05-12:
It looks like (at least in this case) I can code my way around this - I was already guarding accesses to the `r_valids` if `num_stages==0`, but it makes code a bit uglier to move the instantiations around. 

I'd make this low priority, but still something I'd like to see.

ducky64, 2016-05-12:
I'm still not sure how you plan on making zero width Vecs do what you want to do (combinational passthrough) unless Reg does something super unintuitive on having a zero width element passed in or you add in special case code (which defeats the purpose of this in the first place, and you should probably just write your custom `makePipelineStage(numStages: Int)` or `delay(numCycles: Int)` method that internally instantiates either a Reg or Wire).

ucbjrl, 2016-05-13:
At least one other design, (LBNL's OpenSoC) makes similar assumptions (constructing a Vec with zero elements is legal).

ducky64, 2016-05-13:
Can you link to their code (if it's open source) or post a snippet showing how it's used?

aswaterman, 2016-05-13:
Vec(0) should be supported to avoid needless special casing in generators.
It's also presently used in some cases to conditionalize I/O generation.
 (You might say that would better handled with Option or something, but not
supporting Vec(0) is certainly a regression.)

ducky64, 2016-05-13:
I think it would be nice to see some example code, but having Vec(0) be part of an interface definition could make sense, essentially equivalent to Option(None) but more uniform.

zhemao, 2016-07-20:
Is this still an issue for anyone? Vec of size 0 is supported now and we use it extensively in RocketChip's top-level.

terpstra, 2016-07-20:
Ups. I clicked 'close' meaning to close the window. ;-) Probably this issue should be closed anyway.

closed 2016-07-20 by terpstra

seldridge, 2016-07-23:
FYI, it looks like the (to be deprecated?) `Vec.fill` approach doesn't like vectors of size zero. The following craps out: 
```scala
val din = Vec.fill(numReadWritePorts)(UInt(OUTPUT, width = dataWidth))
```
With the following (truncated) error:
```
...
Caused by: java.lang.IllegalArgumentException: requirement failed
	at scala.Predef$.require(Predef.scala:207)
	at chisel3.core.Vec$.do_apply(Aggregate.scala:50)
	at chisel3.core.Vec$.do_fill(Aggregate.scala:95)
...
```
Due to: https://github.com/ucb-bar/chisel3/blob/master/chiselFrontend/src/main/scala/chisel3/core/Aggregate.scala#L50

The supported `Vec(gen: T, n: Int)` works fine, however:
```scala
val din = Vec(UInt(OUTPUT, width = dataWidth), numReadWritePorts)
```

zhemao, 2016-07-27:
No, Vec.fill is supported going forward, but Vec(gen: T, n: Int) is
deprecated.

I think the issue is that you can declare vectors of size 0, but you can't
try to connect anything to them.