chipsalliance/chisel #189: VecLike.read, write is deprecated but not so in Mem

ducky64, 2016-05-12:
Mem extends VecLike. What's the plan here? It seems like apply can take care of the non-masked read/write, so are those planned to be deprecated? Or is the plan to retain Mem.read and Mem.write but eliminate those from Vec/VecLike?

(this has implications for source locator implementation for Mem, because of the inheritance of .read and .write)

aswaterman, 2016-05-13:
I think remove them from Vec/VecLike and keep them for Mem.

sdtwigg, 2016-05-13:
Mem extending VecLike is perhaps a bit 'odd'. VecLike extends IndexedSeq. IndexedSeq has many operations (some subtle, like toString) that will iterate through the entire memory, generating a port for each address.... This is particularly bad because Mem won't cache any of those port creations.

ducky64, 2017-02-06:
Resolution: try to make Mem not extend VecLike and hope all the chisel3 based code doesn't immediately catch fire.
Might keep current behavior in compatibility mode. Might also keep some IndexedSeq operations,

PR #589 (2017-04-22): Remove VecLike/IndexedSeq from Mem type
Resolves #189 

It doesn't seem like any actual uses of Mem relied on VecLike behavior (which is to be expected - a SRAM with ridiculous ports is pretty much useless), so this is a short, sweet PR.
closed 2017-04-22 by ducky64