chipsalliance/chisel #190: Better propagation of variable names to FIRRTL and beyond

ducky64, 2016-05-13:
Currently, only `val`s defined in a Module have their names passed to FIRRTL. This is because of reflection, only those vals are turned into members of the Module and can be accessed by reflection. Obviously, this is bad for debugging, and a lot of `T_*`s get generated.

Source locators (hopefully coming soon!) helps alleviate the problem somewhat by eliminating the guesswork in which line of code generated what, however it still requires a level of indirection (read FIRRTL / Verilog, find original Scala line, read original Scala line and try to map what is happening).

A possible implementation for this may be with macro annotations (http://docs.scala-lang.org/overviews/macros/annotations.html) which seem to be able to expand arbitrary definitions. However, at least as of Scala 2.12, they are still part of a separate macro paradise plugin.

Thoughts?

aswaterman, 2016-05-13:
IMO, the dependence on macro paradise is probably justified by the substantial increase in usability.

sdtwigg, 2016-05-13:
More advanced macro use would be highly preferable to writing a full-scale compiler plugin (or, related work to preprocess the scala files)

colinschmidt, 2016-06-02:
In addition to difficulty debugging having most instances be T_* makes the VLSI flow nearly impossible to use while RTL is changing.

This is a regression from Chisel2 where things like RocketTile would be named as such rather than T_*

PR #236 (2016-07-20): Generate better names for nodes (#190)
For Chisel nodes defined in Module class-level values of type Option or
Iterable, we can still use reflection to assign names based on the name
of the value. This works for arbitrary nesting of Option and Iterable so
long as the innermost type is HasId. Note that this excludes Maps which
always have an innermost type of Tuple2[\_,\_].

Does not fix #190 but does improve things a little.
closed 2016-07-20 by aswaterman