chipsalliance/chisel #192: Curious behavior when wrapping Module instantiation in multiple Module.apply invocations

jackkoenig, 2016-05-17:
The code below seems like a very reasonable thing to write:
```scala
class Minuser extends Module {
  val io = new Bundle {
    val a = UInt(width = 32).flip()
    val b = UInt(width = 32).flip()
    val c = UInt(width = 32)

    c := a -% b
  }
}

object Driver extends App {
  val s = Chisel.Driver.emit(() => Module(new Minuser))
  println(s)
}
```
but Chisel.Driver.emit calls elaborate which invokes Module.apply, so this is essentially double calling the apply method.

This results in the following, somewhat surprising FIRRTL:
```
circuit Minuser : 
  module Minuser : 
    input clk_7 : Clock
    input reset_8 : UInt<1>
    output io_9 : {flip a_3 : UInt<32>, flip b_2 : UInt<32>, c_1 : UInt<32>}
    
    io_9 is invalid
    io_9 is invalid
    node T_5 = sub(io_9.a_3, io_9.b_2)
    node T_6 = tail(T_5, 1)
    io_9.c_1 <= T_6
```

Is there any way we can deal with this?

aswaterman, 2016-05-17:
Since code inside of bundles isn't kosher,
I want to double-check that isn't part of the issue.

chick, 2016-05-17:
The code in the bundle was a typo (would be nice if chisel could flag this somehow).  But by itself, without the extraneous wrapper module, does not cause the problem.

dt27182, 2016-05-25:
I've run into this issue before too. The unexpected names end up producing some unexpected verilog. This either needs to be supported or caught as an exception because otherwise it is a silent error.

commit 952473bd (2016-11-11): Add checks for forgetting Module() wraps or double Module() wrapping

Implemented by changing currentModule to moduleStack
Fixes #192
commit a5f99430 (2016-11-11): Add checks for forgetting Module() wraps or double Module() wrapping

Implemented by changing currentModule to moduleStack
Fixes #192
commit c53543d7 (2016-11-11): Add checks for misuse or omission of Module()

Implemented by adding a Boolean to check for alternating invocations of object
Module.apply and the constructor of abstract class Module.
Fixes #192
PR #362 (2016-11-14): Add checks for forgetting Module() wraps or double Module() wrapping
Implemented by changing currentModule to moduleStack
Fixes #192 

Deleted DeqIOSpec because it wasn't really doing anything and it was missing a Module() wrap anyway. When I added the Module() wrap the test just times out.
closed 2016-11-14 by jackkoenig