chipsalliance/chisel #206: Reg() should support multiple clocks

ben-k, 2016-06-02:
Reg() cannot currently instantiate with a clock other than the default.  In Reg.scala:

`    val clock = Node(x._parent.get.clock) // TODO multi-clock
`

We need this feature for upcoming tapeout.

ben-k, 2016-08-03:
So for now we're planning to just wrap any bare `Reg` in a module call to work around this.  Going forward, though, either this should be implemented or the "TODO" comment should be removed.

grebe, 2016-10-20:
@stevobailey and I would like to bump this

aswaterman, 2016-10-24:
We have also been using the module work-around @ben-k mentioned.

The API for this deserves discussion.  Adding another parameter clock is viable, but do we do the same for RegEnable and RegReset, too?  Or only on Reg, since the multi-clock case is uncommon?

Another strategy entirely is to distinguish clock domains from modules by having clock scopes, something like

    withClock (myOtherClock) {
      val reg = Reg(UInt()) // this is clocked by myOtherClock
    }

This is more composable: if you write a function that generates registers, it will be affected by the surrounding withClock.  Otherwise, you'd have to resort to the module trick again.

colinschmidt, 2016-10-24:
I like the withClock construct.

mwachs5, 2016-10-24:
If we add a withClock construct, do we need a corresponding "withReset"?

aswaterman, 2016-10-24:
Yeah (and probably withClockAndReset as a convenience).

ducky64, 2016-10-24:
Possibly also worth deprecating the old Module clock/reset override constructors in favor of this?
Those seem to need support from the class extending Module to support clock reset overrides (so interface isn't consistent across all Modules).

mwachs5, 2016-10-24:
Agreed with @ducky64. I find myself using a lot of the:

`
myModule.clock = someOtherClock `

rather than passing it in the constructor.

mwachs5, 2016-10-24:
Extending this further than this issue's scope... would I be allowed to do

```
withClockAndReset (myOtherClock, myOtherReset) {
 val foo = Module (new Foo())
}
```

to also override module's clocks and resets (even though this is not necessary, just wondering what this construct would do)

aswaterman, 2016-10-24:
that's what I'm thinking.

ben-k, 2016-10-24:
I strongly support the proposal by @aswaterman - it would alleviate a lot of the current pain points of writing multiclock Chisel3.  In addition to `Reg`, there are several Chisel3 widgets (e.g., `Counter`) that do not extend `module` and so currently lack multiclock utility without an extra wrapper.

aswaterman, 2016-10-24:
OK - I'll try to implement soon.

terpstra, 2016-10-26:
I think this API seems quite ok.

However, I would not add 'withClock' and 'withRest'. There should only be 'withClockAndReset'. If you want to use the same reset, you can trivially pass it. Since you want the reset desertion to be synchronous with the clock, providing a combined method hopefully encourages proper usage.

terpstra, 2016-10-26:
How does this interact with [Seq]Mem?

aswaterman, 2016-10-26:
Agreed re: withClockAndReset only.

For [Seq]Mem, for which clocks are associated with ports, not the storage element itself, withClockAndReset will take effect for any ports created within.

ducky64, 2016-11-22:
Any movement on the implementation of this? I would like to be able to use this too.
@aswaterman 
(if it's not projected within the next week or two, I can make an attempt at implementing this)

aswaterman, 2016-11-22:
@ducky64 I started but got sidetracked by the deadline of the RISC-V workshop, which is next week.  I plan to finish it up after that, i.e., end of next week.

In the mean time, I'd just declare a Module and pass its constructor a clock and reset.

aswaterman, 2016-12-14:
@jackkoenig will take care of the implementation

jackkoenig, 2017-02-16:
Closed via https://github.com/ucb-bar/chisel3/pull/509!!!

closed 2017-02-16 by jackkoenig