chipsalliance/chisel #208: Publish Chisel3 sbt subprojects in a single jar

ucbjrl, 2016-06-03:
We'd like to publish all Chisel3 sbt subprojects in a single jar. It appeared that the initial attempt at this was working, but it was probably masked by the fact that the separately published jars were still around.
See #194

ducky64, 2016-11-21:
This seems to be the case now.

ucbjrl, 2016-11-21:
I disagree. We're still publishing 3 separate jars: coremacros,
chiselfrontend, and chisel3 with sbt 013.11

ducky64, 2016-11-21:
The projects are being published separately, but it seems chisel3 includes everything that is in chiselfrontend and coremacros

ucbjrl, 2016-11-21:
The issue should be updated to "avoid publishing the coremacros and
chiselFrontend jars" since these no longer seem to be required.

ducky64, 2016-11-22:
Hopefully this issue is _actually_ resolved now.

closed 2016-11-22 by ducky64