chipsalliance/chisel #210: BlackBox elaboration fails when io has Bundle element

zhemao, 2016-06-04:
The following Chisel code causes an internal exception.

    class BlackBoxWithBundle extends BlackBox {
      val io = new Bundle {
        val foo = new Bundle {
          val bar = Bool(INPUT)
        }
      }
    }

The exception message is the following.

    java.util.NoSuchElementException: None.get
        at scala.None$.get(Option.scala:347)
        at scala.None$.get(Option.scala:345)
        at Chisel.internal.HasId$class.getRef(Builder.scala:82)
        at Chisel.Data.getRef(Data.scala:35)
        at Chisel.Bundle.Chisel$Bundle$$eltPort$1(Aggregate.scala:333)
        at Chisel.Bundle$$anonfun$toType$1.apply(Aggregate.scala:335)
        at Chisel.Bundle$$anonfun$toType$1.apply(Aggregate.scala:335)
        at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
        at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
        at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
        at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
        at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
        at scala.collection.AbstractTraversable.map(Traversable.scala:104)
        at Chisel.Bundle.toType(Aggregate.scala:335)

This only appears to happen when a nested bundle is used. Vecs do not have the same issue.

aswaterman, 2016-06-16:
does this happen with an identical test using Module instead of BlackBox?

zhemao, 2016-06-20:
You mean if I make a module with nothing inside of it? Let me try it and see.

closed 2016-06-20 by aswaterman