chipsalliance/chisel #217: access to Bundle elements

ucbjrl, 2016-06-09:
The testers (via IOAccessor) need access to Bundle elements as types (so the hwio testers can find the ready/valid signals which gate stepping through the input/output test vectors). As of commit 07fa5622ccc995f925d6d967d2a386540c9064cc namedElts is private[core]. The expedient fix is to make this private[chisel], but we should discuss what should be the "right" solution.

(This is a typical problem. Testers/debuggers do have valid reasons for reaching behind the public API.)

ducky64, 2016-06-09:
Testers / debuggers have reasons to reach beyond the _hardware construction_ API, but it should still use a _public_ API, reasoning being that someone who's not developing Chisel core may also want to write testers / debuggers, and the necessary functions should be exposed to them.

ducky64, 2016-06-09:
Also, it seems like hardware construction time reflection is currently used (I heard one Chisel project apparently queries IO pins for directionality), so exposing Bundle contents may be part of that. But possibly done in a more structured way than just exposing internal data types.

ucbjrl, 2016-06-09:
@donggyukim points out "elements" is part of the public API.

closed 2016-06-09 by ucbjrl