chipsalliance/chisel #219: Chisel rename fails on case-insensitive filesystems.

ucbjrl, 2016-06-09:
Unfortunately, we can't provide both a "Chisel" package and a "chisel" package on platforms with case-insensitive file systems (i.e., Mac OSX):

    [warn] .../chisel3/src/main/scala/chisel/compatibility.scala:6: Class Chisel.package$ differs only in case from chisel.package$. Such classes will overwrite one another on case-insensitive filesystems.
    [warn] package object Chisel {
    [warn]                ^

Running tests in chisel3 result in:

    [info] Compiling 36 Scala sources to /tmp/jrltmp/sbt/Chisel3/scala-2.11/test-classes...
    [error] .../chisel3/src/test/scala/chiselTests/Assert.scala:7: object testers is not a member of package chisel
    [error] import chisel.testers.BasicTester
    [error]               ^
    [error] .../chisel3/src/test/scala/chiselTests/Assert.scala:8: object util is not a member of package chisel
    [error] import chisel.util._
    [error]               ^
    [error] .../chisel3/src/test/scala/chiselTests/Assert.scala:10: not found: type BasicTester
    [error] class FailingAssertTester() extends BasicTester {
    [error]                                     ^
    [error] .../chisel3/src/test/scala/chiselTests/Assert.scala:11: not found: value Bool
    [error]   assert(Bool(false))
    [error]          ^
    [error] .../chisel3/src/test/scala/chiselTests/Assert.scala:13: not found: value Counter
    [error]   val (_, done) = Counter(!reset, 4)
    [error]                   ^
    [error] .../chisel3/src/test/scala/chiselTests/Assert.scala:13: not found: value reset

We can either:
 - require case-sensitive filesystems for Chisel development
 - eliminate the "Chisel" package and compatibility layer
 - "temporarily" rename the lowercase package name to "chisel3"
 - skip the rename altogether

Although OSX suggests the root volume should be case-insensitive, additional volumes may be configured as case-sensitive. A case-sensitive filesystem is recommended for Android development.

aswaterman, 2016-06-10:
Many software projects fall victim to this, including even the GNU C Library.  We mount case-sensitive file systems to get around that issue.  It is annoying and would hinder uptake to some extent.

Can we handle this another way by putting the two packages in different directory subtrees?

Alternatively, using chisel3 as the package name doesn't sound like a bad idea.

ducky64, 2016-06-10:
I tried putting the Chisel stuff in a different package in the caseFix branch, but that doesn't have seemed to work on my Windows machine. Anyone on a case-insensitive Mac want to try it?

ucbjrl, 2016-06-10:
It appears to compile on an OSX case-insensitive FS without warnings about case sensitivity, but almost all tests fail with:

    [info] Exception encountered when attempting to run a suite with class name: chiselTests.DirectionSpec *** ABORTED ***
    [info]   java.lang.NoClassDefFoundError: chisel/package$
    [info]   at chisel.Driver$$anonfun$elaborate$1.apply(Driver.scala:110)
    [info]   at chisel.Driver$$anonfun$elaborate$1.apply(Driver.scala:110)
    [info]   at chisel.internal.Builder$$anonfun$build$1.apply(Builder.scala:116)
    [info]   at chisel.internal.Builder$$anonfun$build$1.apply(Builder.scala:114)
    [info]   at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
    [info]   at chisel.internal.Builder$.build(Builder.scala:114)
    [info]   at chisel.Driver$.elaborate(Driver.scala:110)
    [info]   at chiselTests.ChiselRunners$class.elaborate(ChiselSpec.scala:23)
    [info]   at chiselTests.ChiselPropSpec.elaborate(ChiselSpec.scala:31)
    [info]   at chiselTests.DirectionSpec$$anonfun$1.apply$mcV$sp(Direction.scala:30)
    [info]   ...

ducky64, 2016-06-10:
Yeah, got the same problem. I'm guessing that regardless of the project structure we give it, the classes get flattened at runtime and still collide, it just gets pushed past where the compiler could warn you about it

chick, 2016-06-11:
Let me place my vote for having chisel having only the lower case form and conversion to chisel3 require fixing up the package names in your project.  There is a growing list of things that have to be done to move to chisel3, this one more does not seem that onerous.  I also think we are using up a lot of time trying to solve nearly intractable problems like this

ducky64, 2016-06-11:
Having users move to lowercase chisel was always part of the plan, but that's not due to happen until the "release". This is mainly "how do we support both until everyone is ready to make the huge jump" (compatibility mode).

Another question is how much work needs to go into compatibility mode vs. how much the user needs to put in (for example, perhaps requiring case-sensitive filesystems during the transition period is acceptable)?

closed 2016-11-21 by ducky64