chipsalliance/chisel #220: Out of Memory Error

da-steve101, 2016-06-15:
Hi,

I implemented a simple module as follows:
```scala
class AdderTree( vecSize : Int ) extends Module {
  val io = new Bundle {
    val numIn = Vec( vecSize, UInt( INPUT ) )
    val numOut = UInt( OUTPUT )
  }
  val vecStage = ArrayBuffer[List[UInt]]()
  vecStage.append( io.numIn.toList )
  while ( vecStage.last.size != 1 ) {
    val newStage = vecStage.last.grouped(2).map( x => {
      if ( x.size == 2 ) x(0) + x(1)
      else x(0)
    }).toList
    vecStage.append( newStage )
  }
  io.numOut := vecStage.last(0)
}

class AdderTreeTester( bitWidth : Int, numsToAdd : List[Int] ) extends BasicTester {
  val dut = Module(new AdderTree( numsToAdd.size))
  dut.io.numIn := Vec(numsToAdd.map(x => UInt(x, bitWidth)))
  val sumCorrect = dut.io.numOut === UInt( numsToAdd.reduce(_+_) )
  assert(sumCorrect)
  stop()
}

class AdderTreeSpec extends ChiselPropSpec {
  property("All numbers should be added correctly by an Adder Tree") {
    forAll( safeUIntN( 10 ) ) {
      case(w: Int, v: List[Int]) => assertTesterPasses{ new AdderTreeTester( w, v ) }
    }
  }
}
```
This is currently failing with java heap out of memory error. I increased it but still crashes after the fifth or so "Elaborating Design ... "
```
- /tmp/AdderTreeTester7976910890955580761/AdderTreeTester.v:156: Verilog $finish
- /tmp/AdderTreeTester7976910890955580761/AdderTreeTester.v:156: Second verilog $finish, exiting
[info] [0.000] Elaborating design...
[info] [0.000] Elaborating design...
[info] [0.000] Elaborating design...
[info] [0.000] Elaborating design...
[info] [0.000] Elaborating design...
[info] AdderTreeSpec:
[info] - All numbers should be added correctly by an Adder Tree *** FAILED ***
[info]   OutOfMemoryError was thrown during property evaluation.
[info]     Message: Java heap space
[info]     Occurred when passed generated values (
[info]       arg0 = (0,List()) // 5 shrinks
[info]     )
```

sdtwigg, 2016-06-15:
Your code is has an error that is manifest by those generated values:
If I pass an empty List to the tester, it will create AdderTree with vecSize = 0.
From that, the while loop will never terminate (until it OOMs since you are appending the results each time).

Stylistically, you should probably first calculate the number of stages you will need and then fold over that range.

closed 2016-06-15 by sdtwigg

da-steve101, 2016-06-15:
Ah ok sure, obvious now you point it out.
Thanks :)