chipsalliance/chisel #230: "Reg(init = UInt(width=10))" generates invalid FIRRTL

palmer-dabbelt, 2016-07-05:
I was helping Jonathan with his RISC-V demo and made a brain-o that resulted in invalid FIRRTL being generated.  The problem comes from initializing a Reg to a UInt without a value.  An example can be found here

https://github.com/ucb-bar/rocket-chip/commit/d756f8b456abf120144815047568047c15f1a805

This is just an error message bug: FIRRTL just said "T_88 undefined", which is hard to track to this problem.

aswaterman, 2016-07-05:
This falls in the general class of problems that isn't straightforward to catch with the current frontend design that blithely emits FIRRTL without understanding what's going on.

palmer-dabbelt, 2016-07-05:
I case anyone is interested, here's the error message I get

````
[INFO] Starting High Form Check
Exception in thread "main" firrtl.passes.CheckHighForm$UndeclaredReferenceException: : [module SmiExample] Reference T_88 is not declared.
        at firrtl.passes.CheckHighForm$.firrtl$passes$CheckHighForm$$checkHighFormE$1(Checks.scala:185)
        at firrtl.passes.CheckHighForm$$anonfun$firrtl$passes$CheckHighForm$$checkHighFormS$1$5.apply(Checks.scala:210)
        at firrtl.passes.CheckHighForm$$anonfun$firrtl$passes$CheckHighForm$$checkHighFormS$1$5.apply(Checks.scala:210)
        at firrtl.Mappers$StmtMagnet$$anon$3.map(Mappers.scala:50)
        at firrtl.Mappers$StmtMap.map(Mappers.scala:95)
        at firrtl.passes.CheckHighForm$.firrtl$passes$CheckHighForm$$checkHighFormS$1(Checks.scala:210)
        at firrtl.passes.CheckHighForm$$anonfun$firrtl$passes$CheckHighForm$$checkHighFormS$1$8.apply(Checks.scala:230)
        at firrtl.passes.CheckHighForm$$anonfun$firrtl$passes$CheckHighForm$$checkHighFormS$1$8.apply(Checks.scala:230)
        at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
        at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
        at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
        at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
        at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
        at scala.collection.AbstractTraversable.map(Traversable.scala:104)
        at firrtl.Mappers$StmtMagnet$$anon$4.map(Mappers.scala:42)
        at firrtl.Mappers$StmtMap.map(Mappers.scala:95)
        at firrtl.passes.CheckHighForm$.firrtl$passes$CheckHighForm$$checkHighFormS$1(Checks.scala:230)
        at firrtl.passes.CheckHighForm$.firrtl$passes$CheckHighForm$$checkHighFormM$1(Checks.scala:246)
        at firrtl.passes.CheckHighForm$$anonfun$run$1.apply(Checks.scala:255)
        at firrtl.passes.CheckHighForm$$anonfun$run$1.apply(Checks.scala:253)
        at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
        at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
        at firrtl.passes.CheckHighForm$.run(Checks.scala:253)
        at firrtl.SimpleRun$$anonfun$1$$anonfun$2.apply(LoweringCompilers.scala:46)
        at firrtl.SimpleRun$$anonfun$1$$anonfun$2.apply(LoweringCompilers.scala:46)
        at firrtl.Utils$.time(Utils.scala:53)
        at firrtl.SimpleRun$$anonfun$1.apply(LoweringCompilers.scala:46)
        at firrtl.SimpleRun$$anonfun$1.apply(LoweringCompilers.scala:44)
        at scala.collection.LinearSeqOptimized$class.foldLeft(LinearSeqOptimized.scala:124)
        at scala.collection.immutable.List.foldLeft(List.scala:84)
        at firrtl.SimpleRun$class.run(LoweringCompilers.scala:43)
        at firrtl.ResolveAndCheck.run(LoweringCompilers.scala:80)
        at firrtl.ResolveAndCheck.execute(LoweringCompilers.scala:94)
        at firrtl.Compiler$$anonfun$compile$1.apply(Compiler.scala:96)
        at firrtl.Compiler$$anonfun$compile$1.apply(Compiler.scala:95)
        at scala.collection.LinearSeqOptimized$class.foldLeft(LinearSeqOptimized.scala:124)
        at scala.collection.immutable.List.foldLeft(List.scala:84)
        at firrtl.Compiler$class.compile(Compiler.scala:95)
        at firrtl.VerilogCompiler.compile(LoweringCompilers.scala:192)
        at firrtl.Driver$.compile(Driver.scala:59)
        at firrtl.Driver$.main(Driver.scala:130)
        at firrtl.Driver.main(Driver.scala)
make: *** [/scratch/palmer.dabbelt/work/jrb-demo/rocket-chip/emulator/generated-src/Top.DefaultConfig.v] Error 1
````

and here's the relevant FIRRTL snippit

````
  module SmiExample : 
    input clk : Clock
    input reset : UInt<1>
    output io : {flip smi : {req : {flip ready : UInt<1>, valid : UInt<1>, bits : {rw : UInt<1>, addr : UInt<10>, data : UInt<64>}}, flip resp : {flip ready : UInt<1>, valid : UInt<1>, bits : UInt<64>}}, iobit : UInt<1>}
      
    io is invalid  
    reg resp_valid : UInt<1>, clk with : (reset => (reset, UInt<1>("h00")))  
    node T_86 = and(io.smi.resp.ready, io.smi.resp.valid) @[Decoupled.scala 16:42] 
    when T_86 : @[SmiExample.scala 19:29] 
      resp_valid <= UInt<1>("h00") @[SmiExample.scala 20:16] 
      skip @[SmiExample.scala 19:29] 
    reg read_addr : UInt, clk with : (reset => (reset, T_88))
    node T_89 = and(io.smi.req.ready, io.smi.req.valid) @[Decoupled.scala 16:42] 
    when T_89 : @[SmiExample.scala 24:28] 
      read_addr <= io.smi.req.bits.addr @[SmiExample.scala 25:15] 
      resp_valid <= UInt<1>("h01") @[SmiExample.scala 26:16] 
      skip @[SmiExample.scala 24:28] 
    node T_92 = eq(resp_valid, UInt<1>("h00")) @[SmiExample.scala 29:23] 
    io.smi.req.ready <= T_92 @[SmiExample.scala 29:20] 
    io.smi.resp.valid <= resp_valid @[SmiExample.scala 30:21] 
    reg mem : UInt<64>[1024], clk
    node T_100 = and(io.smi.req.ready, io.smi.req.valid) @[Decoupled.scala 16:42] 
    node T_101 = and(T_100, io.smi.req.bits.rw) @[SmiExample.scala 34:27] 
    when T_101 : @[SmiExample.scala 34:50] 
      node T_102 = not(io.smi.req.bits.data) @[SmiExample.scala 35:34] 
      mem[io.smi.req.bits.addr] <= T_102 @[SmiExample.scala 35:31] 
      skip @[SmiExample.scala 34:50] 
    io.smi.resp.bits <= mem[read_addr] @[SmiExample.scala 37:20] 
    node T_103 = bits(mem[0], 0, 0) @[SmiExample.scala 39:21] 
    io.iobit <= T_103 @[SmiExample.scala 39:12] 
````

palmer-dabbelt, 2016-07-05:
@aswaterman well, you're welcome to WONTFIX this, I don't really care that much.

aswaterman, 2016-07-05:
That's not what I mean... I was just commenting that it's a manifestation of a bigger problem.

sdtwigg, 2016-07-06:
This class of error is what the Binding system in my mega-PR was specifically designed to catch.

As Andrew noted, this is really hard to fix in the stock frontend since the stock frontend simply has no-to-minimal context for the circuit it is generating. A UInt(width=16) looks identical to a Wire(UInt(width=16)).

ducky64, 2017-02-06:
This should trigger a binding error now.

closed 2017-02-06 by ducky64