chipsalliance/chisel #234: PriorityMux now uses [T <: Bits]; but unable to coax Bundle to [T <: Bits] to approximate old Chisel2 behavior

ccelio, 2016-07-19:
I have code that works in Chisel2, but won't compile in Chisel3 (and no compatibility warning).

Chisel2's PriorityMux uses a `[T <: Data]`, but Chisel3 uses `[T <: Bits]`.  What's the rational behind this, and what's the idiomatic way to achieve the old behavior?

**Error Message**:
````scala
[error] /scratch/celio/boom-chisel3-chip/boom/src/main/scala/tage.scala:220: overloaded method value apply with alternatives:
[error]   [T <: Chisel.Bits](sel: Chisel.Bits, in: Seq[T])T <and>
[error]   [T <: Chisel.Bits](sel: Seq[Chisel.Bool], in: Seq[T])T
[error]  cannot be applied to (scala.collection.immutable.IndexedSeq[Chisel.Bool], scala.collection.immutable.IndexedSeq[boom.TageTableResp])
[error]    val best_prediction_bits = PriorityMux(valids.reverse, predictions.reverse)
[error]                               ^
````

The definition of a TageTableResp:

````scala
class TageTableResp(fetch_width: Int, history_length: Int, index_length: Int, tag_sz: Int)(implicit p: Parameters)                 
   extends BoomBundle()(p)                                                                                                         
{                                                                                                                                  
   val takens  = UInt(width = fetch_width)  // the actual prediction                                                               
   val index   = UInt(width = index_length) // the index of the prediction                                                         
   val tag     = UInt(width = tag_sz)       // the tag we computed for the prediction                                                                                                                                                              

   override def cloneType: this.type = new TageTableResp(fetch_width, history_length, index_length, tag_sz).asInstanceOf[this.type]
}                                                                                                                                  
                                                                                                                                   
````

I'm using commit 0ad47ff929aa084d6aff4d378b32968ef19b97c7, but I don't think this is different from current master.

aswaterman, 2016-07-19:
This was resolved by 3b10267257de7662abbbc235d9bfd8a8b89f69f5

closed 2016-07-19 by aswaterman

ccelio, 2016-07-19:
Thanks, I guess I was on the wrong commit on github when I was looking at Mux.scala.