chipsalliance/chisel #235: Difficulty coercing UInt to BitPat for rocket.DecodeLogic() function

ccelio, 2016-07-20:
My adventures porting to Chisel3 continue... (this is on chisel3 master branch now).

I'm getting an error message that I'm finding a bit confusing; it complains that a UInt is not a BitPat type, but I'm fairly confident I'm matching the existing rocket idecode.scala idioms (https://github.com/ucb-bar/rocket/blob/master/src/main/scala/idecode.scala#L45), which are using UInts as well.

**The Error Message**:

````scala
[info] Compiling 43 Scala sources to /scratch/celio/boom-chip/target/scala-2.11/classes...
[error] /scratch/celio/boom-chip/boom/src/main/scala/decode.scala:34: type mismatch;
[error]  found   : chisel3.core.UInt
[error]  required: Chisel.BitPat
[error]     (which expands to)  chisel3.util.BitPat
[error]               List(N, N, X, uopX    , FU_X   ,RT_X,BitPat.DC(2),BitPat.DC(2),X,IS_X,X,X,X,X,N, M_X,   MSK_X,  BitPat.DC(2),        X, X, X, X, N, X, CSR.X)
[error]                                               ^
````    

Line 34 in decode.scala:

````scala
val decode_default: List[BitPat] =                                                                                     
            List(N, N, X, uopX    , FU_X   ,RT_X,BitPat.DC(2),BitPat.DC(2),X,IS_X,X,X,X,X,N, M_X,   MSK_X,  BitPat.DC(2),        X, X, X, X, N, X, CSR.X)
                                                                                                                        

````

`RT_X` is defined as `val RT_X     = UInt(2, 2)`.

Any ideas on what I might be doing wrong? Is there a way to catch this in compatibility mode? Can the error message be clearer?

aswaterman, 2016-07-20:
Rocket has an implicit conversion from BitPat to UInt that handles this (see util.scala); it's not in Chisel proper.

ccelio, 2016-07-20:
Thanks, are there any discussions about implicit conversions being in Chisel proper? Or is it just too painful to handle the operand ordering issues that arise from that? 

Going from Bool->BitPat and UInt->BitPat seems like something that should be default.

closed 2016-07-20 by ccelio

aswaterman, 2016-07-20:
In this particular case, @ducky64 argued against it strongly because the implicit conversion only succeeds on certain BitPats.  I didn't agree because when it fails, it's not silent, but that's where we ended up.