chipsalliance/chisel #241: Vec.fill(n){e} and Vec(n,e) confusion, especially in IO Bundles.

ccelio, 2016-07-21:
I have code that works in Chisel2, and makes it through Chisel3 generation, but errors out early in FIRRTL (actually, it was a FIRRTL compiler exception, but a new fix to FIRRTL catches the error now).

**The code**:

````scala
class ExecutionUnitIO(num_rf_read_ports: Int                                             
                     , num_rf_write_ports: Int                                           
                     , num_bypass_ports: Int                                             
                     , data_width: Int                                                   
                     )(implicit p: Parameters) extends BoomBundle()(p)                   
{                                                                                                        
   val resp    = Vec.fill(num_rf_write_ports) { new DecoupledIO(new ExeUnitResp(data_width)) }   
   // and the fix, commented out...
   // val resp    = Vec(num_rf_write_ports, new DecoupledIO(new ExeUnitResp(data_width)))   
````

**The error:**
````
[INFO] Starting Chirrtl Check
Exception in thread "main" firrtl.passes.PassExceptions: 
firrtl.passes.CheckChirrtl$UndeclaredReferenceException$1:  @[execute.scala 52:47]: 
  [module ALUExeUnit] Reference T_829 is not declared.
````

**The fix**:

Switch from Vec.fill(n){e} to Vec(n,e).

**The request:**

I think it is mandatory that this be caught in Chisel3 before making it to FIRRTL.

It'd also be great if this was caught in the compatibility checker in Chisel2.

da-steve101, 2016-07-22:
I think this is supposed to work in Chisel3. See #104 and #109 where Vec(n,e) was patched but as a "Temporary fix" and hence wasn't extended to fill.

jackkoenig, 2016-09-07:
This error is still occurring with current master

ducky64, 2017-02-06:
Superseded by #445, we will explicitly separate chisel type and hardware constructors.

closed 2017-02-06 by ducky64