chipsalliance/chisel #242: CheckGenders pass in FIRRTL fails

ccelio, 2016-07-21:
Here is code that worked in Chisel2 and is now generating a FIRRTL error from code generated by Chisel3. 

**The error:**
````
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 535:26]: [module BOOMCore]  
  Expression regfile.io.read_ports is used as a sink but can only be used as a source.
````

**The code:**
````scala
// core.scala 535:
regfile.io.read_ports <> register_read.io.rf_read_ports 
````

Regfile code:
````scala
class RegisterFileReadPortIO(addr_width: Int, data_width: Int)(implicit p: Parameters) extends BoomBundle()(p)
{
   val addr = UInt(INPUT, addr_width)                                                                           
   val data = Bits(OUTPUT, data_width)                                                                          
   override def cloneType = new RegisterFileReadPortIO(addr_width, data_width)(p).asInstanceOf[this.type]       
}                                                                                                               
                                                                                                          
                                                                                                                
class RegisterFile( num_registers: Int                                                                          
                  , num_read_ports: Int                                                                         
                  , num_write_ports: Int                                                                        
                  , register_width: Int                                                                         
                  , enable_bypassing: Boolean)                                                                  
                  (implicit p: Parameters) extends BoomModule()(p)                                              
{                                                                                                               
   val io = new BoomBundle()(p)                                                                                 
   {                                                                                                            
      val read_ports = 
         Vec(num_read_ports, new RegisterFileReadPortIO(PREG_SZ, register_width))                 
      ...
   }                                                                                                            

````

Registerread code:

````scala
class RegisterReadIO(                                                                                      
   ...                                                                  
)(implicit p: Parameters) extends  BoomBundle()(p)                                                         
{                                                                                                          
   ...                       
   // interface with register file's read ports                                                            
   val rf_read_ports = 
         Vec(num_total_read_ports, new RegisterFileReadPortIO(PREG_SZ, register_width).flip)                                                                                                         
````

**Question:**

Why is there some sort of implicit directionality on the bulk-connect? It's a bulk-connect! There are inputs and outputs going in both directions!

If I swap the order, I get the same error, but complaining about the other guy being the sink:

````
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 535:35]: [module BOOMCore]  
  Expression register_read.io.rf_read_ports is used as a sink but can only be used as a source.
````

ccelio, 2016-07-21:
The resulting FIRRTL code:

````scala
register_read.io.rf_read_ports <= regfile.io.read_ports @[core.scala 535:35] 
````

````scala
module RegisterFile :                                                 
  input clk : Clock                                                   
  input reset : UInt<1>                                               
  output io : {read_ports : {flip addr : UInt<7>, data : UInt<65>}[5],
````

````scala
module RegisterRead :
  input clk : Clock
  input reset : UInt<1>
  output io : {flip iss_valids : UInt<1>[2], iss_uops :
  ...
  rf_read_ports : { flip addr : UInt<7>, data : UInt<65>}[5],
  ...
````

ccelio, 2016-07-21:
The error appears to be Chisel treating the following two lines of code differently:

````scala
// incorrect
val rf_read_ports = 
         Vec(num_total_read_ports, new RegisterFileReadPortIO(PREG_SZ, register_width).flip) 

// and...

//correct!
val rf_read_ports = 
         Vec(num_total_read_ports, new RegisterFileReadPortIO(PREG_SZ, register_width)).flip 
````

One is flipping the element within the Vec, and the other is flipping the Vec itself. 

Either this should be supported or it is an error that must be caught much earlier.

sdtwigg, 2016-07-22:
This is a duplicate of https://github.com/ucb-bar/chisel3/issues/148

closed 2017-02-06 by ducky64