chipsalliance/chisel #243: Order matters when using bulk-connect between Vecs?

ccelio, 2016-07-21:
Related to the code involved in Issue #242:

There is a second, very unintuitive error involved with the Vec()'s being strict connected: **the order of the Aggregates being bulk-connected matters**. 

Once I have addressed specifying the `flip` on the Vec, and not the elements within the Vec (to solve #242), I run into this:

Error:
````
regfile.io.read_ports <> register_read.io.rf_read_ports 
````

No error:
````
register_read.io.rf_read_ports <> regfile.io.read_ports
````

This is nonsensical. Both sides have inputs and outputs. I tried modifying Aggregate.scala in chisel3.core to attempt to `bulkConnect` Vec's, and I got it to use a different FIRRTL connector, but the error remained:

````
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 535:26]: [module BOOMCore]  
  Expression regfile.io.read_ports is used as a sink but can only be used as a source.
firrtl.passes.CheckGenders$WrongGender:  @[core.scala 535:26]: [module BOOMCore]  
  Expression register_read.io.rf_read_ports is used as a source but can only be used as a sink.
````

FIRRTL code (after I modified Aggregate.scala):
````
regfile.io.read_ports <- register_read.io.rf_read_ports @[core.scala 535:26]
````

aswaterman, 2016-07-22:
The nonsensical thing, IMO, is the name of the operator.  <> suggests commutativity.  Maybe <- would be better.

ccelio, 2016-07-22:
It is nonsensical because **both** sides have inputs and outputs. There is no directionality at the Bundle level, only at the element level.

aswaterman, 2016-07-22:
Is it not a FIRRTL issue, then?

ccelio, 2016-07-22:
There is no good way for me to distinguish what is a FIRRTL issue in handling Chisel code, or a Chisel issue in generating malformed FIRRTL.  I leave that up for someone knowledgeable of both (such as yourself) to decide if it should be moved.

aswaterman, 2016-07-22:
It's not obvious to me either, but can you provide the declarations of the offending signals so @azidar can give some insight?

ducky64, 2017-02-06:
This is a compatibility layer issue that isn't a problem with `chisel3._`.

closed 2017-02-06 by ducky64