chipsalliance/chisel #244: Missing sub-word assignment- breaks existing Chisel2 code of bit vectors.

ccelio, 2016-07-22:
I have lots of Chisel2 code that is written using sub-word assignments. Instead of a Reg(Vec.fill(n){Bool()}) I used a Reg(Bits(width=n)).  

There is no compatability warning and no error in Chisel3. Instead, Chisel3 generates broken FIRRTL that creates a temporary node for the bit that is read, and then attempts to assign to this "source".

**My code:**

````scala
val table_bsy = Reg(init=Bits(0,PHYS_REG_COUNT))           
                                                           
for (wb_idx <- 0 until num_wb_ports)                       
{                                                          
   when (io.unbusy_pdst(wb_idx).valid)                     
   {                                                       
      table_bsy(io.unbusy_pdst(wb_idx).bits) := NOT_BUSY   
   }                                                       
}                                                          
````

Broken FIRRTL code:
````
when io.unbusy_pdst[0].valid : @[rename.scala 392:7]                         
  node T_111 = dshr(table_bsy, io.unbusy_pdst[0].bits) @[rename.scala 393:19]
  node T_112 = bits(T_111, 0, 0) @[rename.scala 393:19]                      
  T_112 <= UInt<1>("h00") @[rename.scala 393:49]                             
  skip @[rename.scala 392:7]                                                 
````

aswaterman, 2016-07-22:
I don't think there's any plan to support subword assignment.  This is another issue in the long list of things like #230, where Chisel should, but doesn't, catch the error before emitting FIRRTL.

ccelio, 2016-07-22:
> I don't think there's any plan to support subword assignment. 

So subword assignment originally developed out performance issues with Vec of Bools. I take it we're moving back to Vec of Bools, and hoping the FIRRTL passes figure it out?

sdtwigg, 2016-07-22:
I think currently firrtl LowerTypes will explode all Vecs into their constituent elements.

aswaterman, 2016-07-22:
@jackkoenig and I talked about the possibility of optimizing Vec[Bool] in FIRRTL in the future.  Still seems like a good plan.

In many cases, using UInt and the bitSet method can simulate the desired effect without fracturing into Bools.

ccelio, 2016-07-22:
> In many cases, using UInt and the bitSet method can simulate the desired effect without fracturing into Bools.

Ah, and this is currently supported? I'll have to give that a try then.

aswaterman, 2016-07-22:
It doesn't mutate the state. It just returns a new word with the bit set or
clear. That's why it is only sometimes sufficient.

ducky64, 2017-02-06:
No longer supported in chisel3. Suggested alternative to to cast to a Vec or Bundle and cast back.

(most uses of subword assignment is because of inadequately designed data structures)

closed 2017-02-06 by ducky64