chipsalliance/chisel #245: Vector of Queues wants a cloneType for QueueIO

seldridge, 2016-07-24:
Chisel3 doesn't like a Vector of Queues, e.g.:
```scala
class QueueBundle extends Bundle {
  val x = Bool
}

class VecOfQueues extends Module {
  val io = new Bundle {
    val x = Bool(INPUT)
    val y = Bool(OUTPUT)
  }

  val voq = Vec(Module(new Queue(new QueueBundle, 2)).io, 2)
}
```
The above example errors out complaining about `QueueIO` needing a `cloneType` method:
```
[error] (run-main-2) chisel3.internal.ChiselException: [error] VecOfQueues.scala:15: Parameterized Bundle class chisel3.util.QueueIO needs cloneType method in class VecOfQueues
[error] [error] VecOfQueues.scala:15: Parameterized Bundle class chisel3.util.QueueIO needs cloneType method in class VecOfQueues
[error] [error] VecOfQueues.scala:15: Parameterized Bundle class chisel3.util.QueueIO needs cloneType method in class VecOfQueues
[error] [error] VecOfQueues.scala:15: Parameterized Bundle class chisel3.util.QueueIO needs cloneType method in class VecOfQueues
[error] CODE HAS 4 ERRORS and 0 WARNINGS
```

Giving `QueueIO` a dedicated `cloneType` method (like `DeqIO` and `EnqIO`) does get it past this, but I can't comment as to whether or not this is the correct approach. I can throw a pull request your way if you want to use it.

sdtwigg, 2016-07-24:
You should be aware that you are creating a Vec[QueueIO] that has no relation to the original Queue, NOT creating a couple of Queues and then exposing a Vec[QueueIO] where each element points to the io of a Queue. (In fact, the created Queue is basically lost.) It would be equivalent for you to write:
val voq = Vec(new QueueIO(new QueueBundle, 2)), 2)

Further, (and I would want to verify this but am pretty sure this is right): QueueBundle actually contains no Data subfields. "val x = Bool" is assigning x to the 'Bool companion object', which is not a data but merely a container holding factory methods for creating Bool

seldridge, 2016-07-24:
Yes, I'm unfortunately aware... I'm coming (somewhat painfully) from Chisel2 and I thought that this was the advised way to create a vector of modules (you naturally can't do the naive `Vec(Module)` because `Module` violates the `[T <: Data]` bounds). Similarly, the old Chisel2 FAQ still lists this as the suggested approach: https://chisel.eecs.berkeley.edu/faq.html. 

You hit on a critical point in that creating modules this way causes the module to get "swallowed" and lose accessibliity from Chisel/Scala. For example, you lose the ability to loop over the created modules and run some information printf-like methods.

Is there a better way to go about this module creationg for Chisel3?

I'm not sure how to best attack your second point. I may have pulled out too much with this MWE... The original example is a mess with all sorts of inherited nonsense. FWIW, the following still elicits the error without the fix:
```scala
class QueueBundle extends Bundle {                                                                                                               
  val x = Bool()                                                                                                                                 
  val y = UInt(width = 10)                                                                                                                       
}  
```

jackkoenig, 2016-07-26:
Related discussion has continued in #247 but I thought I'd respond with this here

While it is still often advisable to use the standard Vec of io approach to creating several modules, it is also totally acceptable to use Scala datastructures to do the same. Previously, Modules created in this way would get annoying temporary instance names, but this was recently fixed such that so long as you use Iterable\[T <: Data\] (arbitrary nestings are fine, as are Options) and assign it to a class level public val, you will get sensible names. 

Consider the following:
```scala
class QueueBundle extends Bundle {
  val x = Bool()
}

class SeqOfQueues extends Module {
  val io = new Bundle {
    val x = Bool(INPUT)
    val y = Bool(OUTPUT)
  }

  val soq = Seq.fill(2)(Module(new Queue(new QueueBundle, 2)))
}
```

This results in the following firrtl for SeqOfQueues:
```
  module SeqOfQueues : 
    input clk : Clock
    input reset : UInt<1>
    output io : {flip x : UInt<1>, y : UInt<1>}
    
    io is invalid
    inst soq_0 of Queue @[Driver.scala 19:31] 
    soq_0.io is invalid
    soq_0.clk <= clk
    soq_0.reset <= reset
    inst soq_1 of Queue @[Driver.scala 19:31] 
    soq_1.io is invalid
    soq_1.clk <= clk
    soq_1.reset <= reset
```

You can use the seq as you would any Scala data structure.

seldridge, 2016-07-26:
Ah, that's actually incredibly useful! Interestingly, the name generation for Scala `Seq` avoids some of the odd underscore names that Chisel `Vec` likes to produce. This further avoids the problem of losing direct access to the created Modules.

@jackkoenig -- maybe you could comment on the reasons for the advisability vs. acceptability of the Chisel `Vec` vs. Scala `Seq`?

jackkoenig, 2016-07-26:
I think the weird names coming from the Vec is actually a bug, I'll file an issue.

So hopefully someone more knowledgeable will correct me if I'm wrong, but here is my take:

Vecs are hardware datatypes and thus used for that purpose, the two examples that come to mind are:    
1. Custom Datatypes - You use the type form of Vec in custom Bundles, it doesn't make sense to put a Seq there (and it doesn't work)
2.  Dynamic indexing - if you want a hardware index into a hardware array, a Vec is the right choice

For non-hardware things, I tend to use Scala datastructures. When I'm trying to generate lots of modules, I'm not trying to create a hardware datatype, so I will generally use Seqs. However, say I have a group of modules and I want to dynamically index their ios: I could create the Modules in a Seq then create a Vec of their ios, but I also might just create them in a Vec and return the ios. I don't think I'm super consistent on which route I take.

Conveniently, if you have something of type Vec[T <: Data], you can actually connect to it from Seq[T], but I'm not sure how far that extends.

seldridge, 2016-10-12:
As I'm back to hitting this again. The problem is characterized by the following (comparing a vector of queues vs a sequence of queues):
```scala 
import Chisel._
import chisel3.util.Queue

class QueueBundle extends Bundle {
  val x = Bool()
}

class VecOfQueues extends Module {
  val io = new Bundle {}
  val voq = Vec.fill(2)(Module(new Queue(new QueueBundle, 2)).io)
}

class SeqOfQueues extends Module {
  val io = new Bundle {}
  val soq = Seq.fill(2)(Module(new Queue(new QueueBundle, 2)))
}
```
VecOfQueues fails with: 
```
chisel3.core.Binding$BindingException: : Already bound to PortBinding(chisel3.util.Queue@8,Some(output))
```
SeqOfQueues works fine.

I have not had a chance to dig around and see what's going on.

Note: PR #246 is no longer any help here due to some refactoring in `Decoupled.scala`.

This is off of bd78bb4ae8f7f6efd0abc76cb7942739de6b2da1.

seldridge, 2016-10-12:
I still need to make sure this is right, but I think `class QueueIO` just needs an `override def cloneType`, i.e., [Line 92](https://github.com/seldridge/chisel3/blob/5982dd8465a289f9b7b4bcb0a1bdccb48dc0cb41/src/main/scala/chisel3/util/Decoupled.scala#L92) of #246.

That seems to be giving me the correct behavior, but I'll look at it more.

Without this, I attempted to create a `Wire` of `Vec` of `QueueIO` and then wire this up manually to the `Seq` of `Queue`, but was getting similar failures just for the `Vec` of `QueueIO`.

sdtwigg, 2016-10-12:
I think I know what this is: Related to my comments in #300 

If you don't define a cloneType, Chisel will attempt to use some reflection magic to create one for you. If this fails, it simply registers an error and then returns this. However, 'this' isn't fresh so you get a slew of other errors. Eventually one of them just excepts out immediately. As per that prior discussion, cloneType shouldn't be returning this when it fails as it leads to further corruption.

PR #246 (2016-10-24): Add a cloneType method to QueueIO
Fixes #245.
closed 2016-10-24 by ucbjrl