chipsalliance/chisel #247: Vec.fill is call-by-name while Vec.apply is call-by-value?

seldridge, 2016-07-25:
The differences between the `fill` and `apply` methods of `Vec` have created some confusion for me. Specifically `fill` is call-by-name, while `apply` is call-by-value (https://github.com/ucb-bar/chisel3/blob/master/chiselFrontend/src/main/scala/chisel3/core/Aggregate.scala#L27, https://github.com/ucb-bar/chisel3/blob/master/chiselFrontend/src/main/scala/chisel3/core/Aggregate.scala#L92). One of the results of this is that `fill` can be used to instantiate a vector of modules, but `apply` cannot.

This conflicts with Chisel2 where both methods are call-by-name (https://github.com/ucb-bar/chisel/blob/master/src/main/scala/Vec.scala#L52, https://github.com/ucb-bar/chisel/blob/master/src/main/scala/Vec.scala#L86).

Is there a reason for this? This seems highly unintuitive to me as I generally view `fill` and `apply` (and `tabulate`, etc.) as all equivalent in terms of behavior, however, this introduces an odd disconnect.

I suspect this is the root cause of #148 and #242 which both use `Vec.apply`.

An overly verbose MWE:
```scala
import chisel3._
import chisel3.util._

class PassThroughBundle extends Bundle {
  val x = Bool(INPUT)
  val y = Bool(OUTPUT)
}

class PassThrough extends Module {
  val io = new PassThroughBundle
  io.y := io.x
}

class VecModNoParamTest extends Module {
  val inside = Vec.fill(2)(Module(new PassThrough).io) // <- This works
  // val inside = Vec(2, Module(new PassThrough).io)   // <- This doesn't!
  val n = inside.length
  val io = new Bundle {
    val data = inside(0).cloneType
    val sel = UInt(width = log2Up(n))
  }

  (0 until n).map(i => inside(i).x := io.data.x)
  io.data.y := inside(io.sel).y
}
```

sdtwigg, 2016-07-25:
So, it is somewhat confusing because Vec contains two types of functions:
1. factory methods for Vecs: Vec.apply(Int, Data) . These just define hardware data structures, but do not actually create hardware directly. You just use these to build up Bundle descriptions, and pass as gen arguments to registers, wires, and other datatypes like Decoupled and Valid.
2. Methods for creating dynamically indexable Wire(Vec(...)): Vec.apply(Seq[Data]), Vec.fill(...), Vec.tabulate(...). These are much different in that they are actually directly creating and returning hardware objects.

The returns of the methods in 1 are generally used as gen parameters whereas the returns of methods in 2 can be used as init parameters. (Note, since the methods in 1 do not actually create hardware objects, they CAN'T be used as init parameters.)

Personally, I'm not a huge fan of the same chisel.core object having two sets of functions with such disparate behavior and use-cases (particularly since apply is especially overloaded); however, that is the current API.

seldridge, 2016-07-25:
Got it. That explanation makes sense.

Is there any way that this can be checked and error out in a predictable way? The resulting error here is pretty opaque unless you've mucked around with passing Chisel Modules/Bundles as parameters, have made the mistake of trying to do that call-by-value, and are inspecting Aggregate.scala.

I also now see what you mentioned in response to #245 (though that does fail for both `fill` or `apply`). Since the `Module(new MyModule).io` is passed call-by-value for the `apply` method, then `apply` sees the `gen.cloneType` of the IO and not the module and generates a bunch of IO with no relation to anything. Passing call-by-name there won't help, as you mention in #248, as this will still be running the `cloneType` of the IO. Or at least I think that's what's going on here... Phew.

For documentation purposes, the specific error is:
```
[error] (run-main-9) firrtl.passes.PassExceptions: 
[error] firrtl.passes.CheckHighForm$UndeclaredReferenceException:  @[Issues.scala 40:36]: [module VecModNoParamTest] Reference inside is not declared.
[error] firrtl.passes.CheckHighForm$UndeclaredReferenceException:  @[Issues.scala 40:36]: [module VecModNoParamTest] Reference inside is not declared.
[error] firrtl.passes.CheckHighForm$UndeclaredReferenceException:  @[Issues.scala 41:13]: [module VecModNoParamTest] Reference inside is not declared.
```

For produced firrtl only generating one instance of `PassThrough` when using `apply`:
```
circuit VecModNoParamTest : 
  module PassThrough : 
    input clk : Clock
    input reset : UInt<1>
    output io : {flip x : UInt<1>, y : UInt<1>}
    
    io is invalid
    io.y <= io.x @[Issues.scala 13:8] 
    
  module VecModNoParamTest : 
    input clk : Clock
    input reset : UInt<1>
    output io : {data : {flip x : UInt<1>, y : UInt<1>}, sel : UInt<1>}
    
    io is invalid
    inst PassThrough_1 of PassThrough @[Issues.scala 34:29] 
    PassThrough_1.io is invalid
    PassThrough_1.clk <= clk
    PassThrough_1.reset <= reset
    inside[0].x <= io.data.x @[Issues.scala 40:36] 
    inside[1].x <= io.data.x @[Issues.scala 40:36] 
    io.data.y <= inside[io.sel].y @[Issues.scala 41:13] 
```

Compared with the correct firrtl when using `fill`:
```
circuit VecModNoParamTest : 
  module PassThrough : 
    input clk : Clock
    input reset : UInt<1>
    output io : {flip x : UInt<1>, y : UInt<1>}
    
    io is invalid
    io.y <= io.x @[Issues.scala 13:8] 
    
  module VecModNoParamTest : 
    input clk : Clock
    input reset : UInt<1>
    output io : {data : {flip x : UInt<1>, y : UInt<1>}, sel : UInt<1>}
    
    io is invalid
    inst PassThrough_2 of PassThrough @[Issues.scala 33:34] 
    PassThrough_2.io is invalid
    PassThrough_2.clk <= clk
    PassThrough_2.reset <= reset
    inst PassThrough_1_1 of PassThrough @[Issues.scala 33:34] 
    PassThrough_1_1.io is invalid
    PassThrough_1_1.clk <= clk
    PassThrough_1_1.reset <= reset
    wire inside : {flip x : UInt<1>, y : UInt<1>}[2] @[Issues.scala 33:27] 
    inside is invalid @[Issues.scala 33:27] 
    inside[0] <- PassThrough_2.io @[Issues.scala 33:27] 
    inside[1] <- PassThrough_1_1.io @[Issues.scala 33:27] 
    inside[0].x <= io.data.x @[Issues.scala 40:36] 
    inside[1].x <= io.data.x @[Issues.scala 40:36] 
    io.data.y <= inside[io.sel].y @[Issues.scala 41:13] 
```

ducky64, 2017-02-06:
Superseded by #445

closed 2017-02-06 by ducky64