chipsalliance/chisel #249: BlackBox Doesn't Like Bulk Connect

seldridge, 2016-07-28:
#### Problem
Bulk connecting a Chisel 3 BlackBox fails on `getRef`. Explicit connections work. I'm currently assuming this is because the emitted BlackBox doesn't have the same interface connections as an equivalent Module. The MWE example below creates a BlackBox and a Module with equivalent internals and then tries to bulk connect them in the same way.

#### MWE
```scala
class MyBlackBoxIO extends Bundle {
  val a = Bool(INPUT)
  val b = Bool(OUTPUT)
}

trait Internals {
  val io = new MyBlackBoxIO
  io.b := io.a
}

class MyModule   extends Module   with Internals
class MyBlackBox extends BlackBox with Internals

class BlackBoxSubModule extends Module {
  val io = new Bundle {
    val data = Vec(2, new MyBlackBoxIO)
  }

  val mod = Module(new MyModule)
  val bbox = Module(new MyBlackBox)

  // Bulk connections
  io.data(0) <> mod.io
  // io.data(1) <> bbox.io              // <-- This fails!

  // Explicit connection of bbox
  // bbox.io.a := io.data(1).a          // <-- These are ok
  // io.data(1).b := bbox.io.b
}
```

#### Output FIRRTL
```
circuit BlackBoxSubModule : 
  module MyModule : 
    input clk : Clock
    input reset : UInt<1>
    output io : {flip a : UInt<1>, b : UInt<1>}
    
    io is invalid
    io.b <= io.a @[BlackBox.scala 12:8] 
    
  extmodule MyBlackBox : 
    output b : UInt<1>
    input a : UInt<1>
    
    
  module BlackBoxSubModule : 
    input clk : Clock
    input reset : UInt<1>
    output io : {data : {flip a : UInt<1>, b : UInt<1>}[2]}
    
    io is invalid
    inst mod of MyModule @[BlackBox.scala 23:19] 
    mod.io is invalid
    mod.clk <= clk
    mod.reset <= reset
    inst bbox of MyBlackBox @[BlackBox.scala 24:20] 
    bbox.b is invalid
    bbox.a is invalid
    io.data[0] <- mod.io @[BlackBox.scala 27:14] 
```

#### Bulk Connect of `bbox`
If I enable the bulk connect on `bbox`:
```scala
  io.data(1) <> bbox.io
```
I get the following truncated error:
```
[error] (run-main-1c) java.util.NoSuchElementException: None.get
java.util.NoSuchElementException: None.get
	at scala.None$.get(Option.scala:347)
	at scala.None$.get(Option.scala:345)
	at chisel3.internal.HasId$class.getRef(Builder.scala:88)
	at chisel3.core.Data.getRef(Data.scala:30)
	at chisel3.internal.firrtl.Node.fullName(IR.scala:53)
	at chisel3.internal.firrtl.Emitter.chisel3$internal$firrtl$Emitter$$emit(Emitter.scala:26)
	at chisel3.internal.firrtl.Emitter$$anonfun$moduleDefn$1$$anonfun$apply$mcV$sp$2.apply(Emitter.scala:76)
	at chisel3.internal.firrtl.Emitter$$anonfun$moduleDefn$1$$anonfun$apply$mcV$sp$2.apply(Emitter.scala:70)
...
```

However, doing a standard connect works:
```scala
  bbox.io.a := io.data(1).a
  io.data(1).b := bbox.io.b
```
Creating the following connections in FIRRTL:
```
    bbox.a <= io.data[1].a @[BlackBox.scala 32:13] 
    io.data[1].b <= bbox.b @[BlackBox.scala 33:16] 
```

#### Other Thoughts
I was surprised to see that the interface of the emitted BlackBox does not match the interface of the equivalent Module. I would anticipate, perhaps naively, that this should be the case. Looking at some old emitted Chisel2 Verilog, BlackBoxes had ports `io_XX` whereas Chisel3 is ignoring the io portion of this. This decision may be intentional? I haven't dug into how the BlackBox is actually emitted to see what's going on here or to think about a fix (assuming this is incorrect behavior).

ben-k, 2016-07-29:
Regarding "Other Thoughts": Removing the mandatory `io_` header for BlackBox IOs was a requested change.  It's impossible to mandate that all IP follows this naming convention, so Chisel should have the flexibility to emit appropriate Verilog wrappers.

seldridge, 2016-07-29:
@ben-k -- that makes perfect sense. I guess it's been over a year since I last had to deal with that, but you're exactly right. I had to wrap all of my underlying BlackBox'd Verilog in Chisel2 compatibility layer `io_` wrapper modules. Requiring that compatibility layer _is_ unnecessary.

seldridge, 2016-07-29:
Judging from [BlackBox.scala](https://github.com/ucb-bar/chisel3/blob/master/chiselFrontend/src/main/scala/chisel3/core/BlackBox.scala#L9), it looks like BlackBox isn't even fully spec'd out, yet. One note, the ordering of the bulk assign doesn't make a difference for the example I provided, though `<>` is just wonky anyway and may be phased out per #162.

Is there any guiding idea on how BlackBox should be handled? I can always take a stab at it, but I expect there have been some internal discussions on how you all want to handle this.

ducky64, 2017-07-25:
Subsumbed by #633 (which addresses the root cause), and I don't think this bug appears anymore because chisel3 now generates leaf-level single-connects instead of bulk-connects.

closed 2017-07-25 by ducky64