chipsalliance/chisel #250: Unconnected I/Os: how should we manage large project collaboration?

ccelio, 2016-08-06:
**TL;DR**: we need a way to tell if an INPUT (that is "important/necessary") is unconnected.

I recognize that many IOs are purposefully left unconnected - which makes building generators easier - but the current situation makes large project collaboration more difficult. 

--------

Chisel has reached a pretty cool point - we have some successful projects that are used across a number of different institutions/universities/companies where code is being used and pushed back to.

Unfortunately, in my situation, I'm continuing to struggle with "API" changes in the Chisel code I am interfacing with. I am instantiating a number of Chisel Modules from other projects which continue to change their I/O bundle (an "API" change, if you will).

- If the name of an IO field changes, I get a compile error (good!).

- If an IO field is removed that I am connected to, I get a compile error (good!).

- If an IO field is added, nothing happens and the input is dangling (uh oh!).

------------

In software development, the equivalent scenario would be changing the inputs to a function.  Adding new arguments can either be provided with a default value (to gracefully handle the update) or an error out when the function call in the user code no longer matches the library's signature.

````scala
def myFunc(a: Int) // old version
def myFunc(a: Int, b: Int=0) // updated version, silently continues to work
def myFunc(a: Int, b: Int) // updated version, fails noisily, assuming myFunc(a) is removed.
````
-----

Regression tests are not enough here:

1) the coverage may not be enough to catch the change. Or the regression may "only" be a harder-to-detect performance change.

2) If regression tests start to fail, it is still incredibly time consuming to track down which (of potentially many) changes is to blame.

Is it not feasible to rely on regression tests, and there is too much churn to make it feasible to rely on upstream change logs or commit logs to manage this issue (I do appreciate this is partly a problem caused by how successful Chisel has been).

I recognize that many IOs are purposefully left unconnected, which makes building generators easier, and any solution must also address this fact.

------------

Questions:

- Is there a way to detect inputs that are unconnected and drive outputs that are connected and provide an error/warning when this happens?

- Is there a way to detect which Inputs are actually Don't-Cares so we don't get hit with false positives to help cope with writing/using parameterized generators? 

- Is there a way to detect there has been an API change?

- Should "library writers" specify which Inputs are required (or perhaps the inverse - which aren't required?).  

------------

Thoughts?

seldridge, 2016-08-06:
Yes, yes, yes, yes, yes. I've been thinking of support along these lines for a while. Most of my frustrating to debug errors originate from some dangling IO (which is rightly randomized) that must, under all possible circumstances be connected but I either forgot to connect it or missed some internal connection. E.g., and naively, if you're using a Valid interface, the valid line cannot be unconnected. Similarly, certain signals in the RoCC interface must be connected (tilelink acquire/grant). But how do I know what these are beyond either a full understanding of the interface or trial and error?

This, I expect, will have to involve FIRRTL annotations and specifically their propagation. A signal with a "this must be connected" annotation needs to propagate that information to any of its connections. Any violation of that annotation is implicitly an error. Unfortunately, I don't think that this is something that Chisel can know when emitting FIRRTL, but should fit with what I've glimpsed of the discussions of FIRRTL annotations (I'm on my phone otherwise I'd link the issue #...).

aswaterman, 2016-08-06:
FIRRTL can detect disconnected instance inputs and disconnected module outputs, but Chisel3 automatically emits the FIRRTL construct "invalid" to indicate that their connection was not necessary.  The intent here was to match Chisel2 behavior to ease porting.

Perhaps the right thing to do is to make this behavior part of the Chisel compatibility layer, so that old code still works, and new code disallows disconnected instance inputs and module outputs, unless explicitly invalidated (to use the FIRRTL terminology).

ducky64, 2017-02-06:
Resolution was to add a isInvalid API so that we can explicitly mark ports as unused and error out on accidentally unused ports.

ducky64, 2017-02-06:
Superseded by #413

closed 2017-02-06 by ducky64