chipsalliance/chisel #253: Wanted: Equality check for Bits (and Bool has a problem with Num[T])

grebe, 2016-08-10:
What I am trying to do is write a tester that works for subclasses of bits, e.g. 
`class SomeTester[T<:Bits](t: T) extends BasicTester {}`
that will construct some objects of type T and later on check that the ones that are supposed to be equal are. I then check that the tester works for UInt, SInt, and Bool.

Unfortunately, there are some problems. Firstly, Bits does not define ===. Num also does not define === (but probably should). However, adding === to Num and using `T<:Bits with Num[T]` still has a problem with Bool.

 Bool extends UInt. This has the side effect that it has trait Num[UInt], which leads to some strangeness. For example, you can compare Bools w/ UInts but not other Bools (not that comparison makes sense for a Bool). Bool probably shouldn't have trait Num in the first place.

I suggest that Bits probably should have some notion of equality, but we don't want bits to have `def ===(that: Bits)`, we want `def ===(that: (SUBCLASS))`. I think the way scala traditionally deals with types in scenarios like this is to have some sort template trait (e.g. SeqLike, TraversableLike, etc.). Is there a reason we don't do this? I imagine this would clean up some of our other uses of this.type too.

sdtwigg, 2016-08-10:
You can work around the Bool inference issue with the following type signatures: 
class SomeTester[A<:B,B<:Data](t: A with Num[B])
or
class SomeTester[T<:Data](t: Num[T])

When t is Bool, the former will resolve to A=Bool,B=UInt and the latter resolves to T=UInt. The latter is simpler. (You can change Data back to Bits if you are actually using Bits-specific functions.).

The scalac typer is somewhat greedy on the matter of type parameter inference and thus doesn't realize that Bool can be relaxed to UInt in order to satisfy the upper bound. It just greedily tries T = Bool and then insists (correctly) that Bool <: Num[Bool] does not hold.

grebe, 2016-08-10:
Yeah. The workaround I had been using for the Bool inference issue was T<:Num[_], but the problem remains that actually using the comparison functions is broken b/c Bool is not Num[Bool]. I had been emulating `a===b` with `a>=b && a<=b`, but if `a`and `b` are both Bool this does not work.

I guess the real problem is that there is no convenient class/trait at the root of UInt/SInt/Bool that has `===`. I just noticed that do_contains in Aggregate has `ev: T <:< UInt` to do this kind of thing, but that feels very hackish to me.

sdtwigg, 2016-08-10:
Sorry, github markdown hid my actual suggested code. I fixed it up with code markers so it wouldn't do that

aswaterman, 2016-08-10:
Do you intend for === to mean value equality or bitwise equality?  Right now, it means both (since it's only defined on UInt/UInt and SInt/SInt).

I agree that Num would ideally define === and =/=.

grebe, 2016-08-10:
I think `===` should mean value equality, and if you want bitwise equality I think you should reinterpret the objects as bits or have some other operator. Reinterpreting is kind of tricky (e.g. comparing SInts with different widths: converting to UInts first won't do the sign extension, so reinterpreted negative SInts won't be equal but positive ones will be if you aren't careful).

Regardless of what `===` means, I think there should be some notion of equality somewhere in the type hierarchy that can be used with generics.

ducky64, 2017-02-06:
Resolution: don't use bits.

If you want to compare bitwise equality between UInt and SInt, do it explicitly.

closed 2017-02-06 by ducky64