chipsalliance/chisel #256: Chisel does not like parameters in modules that are called width

zhemao, 2016-08-12:
This has been cropping up for a while now. Chisel gives really inscrutable error messages if you ever name a parameter to a module or bundle "width". It's easy to work around, but it does cause a lot of confusion when you encounter it. If you try to instantiate a module in Chisel that is declared as follows ...

    class AdjustableWidth(width: Int) extends Module {
      val io = new Bundle {
        val out = UInt(OUTPUT, width=width)
      }

      io.out := UInt(1)
    }

You get the error message

    java.lang.reflect.InvocationTargetException
    Caused by: java.lang.UnsupportedOperationException: empty.reduceLeft
        at scala.collection.TraversableOnce$class.reduceLeft(TraversableOnce.scala:178)
        at scala.collection.mutable.ArrayBuffer.scala$collection$IndexedSeqOptimized$$super$reduceLeft(ArrayBuffer.scala:48)
        at scala.collection.IndexedSeqOptimized$class.reduceLeft(IndexedSeqOptimized.scala:74)
        at scala.collection.mutable.ArrayBuffer.reduceLeft(ArrayBuffer.scala:48)
        at scala.collection.TraversableOnce$class.reduce(TraversableOnce.scala:206)
        at scala.collection.AbstractTraversable.reduce(Traversable.scala:104)
        at chisel3.core.Aggregate.width(Aggregate.scala:19)

However, just changing it by renaming the parameter makes it work just fine.

    class AdjustableWidth(w: Int) extends Module {
      val io = new Bundle {
        val out = UInt(OUTPUT, width=w)
      }

      io.out := UInt(1)
    }

Full example at https://github.com/zhemao/chiselbugs/blob/master/src/main/scala/WidthParameter.scala

zhemao, 2016-08-12:
It seems that the issue is with using a variable called "width" to declared the UInt. The following code works.

    class AdjustableWidth(width: Int) extends Module {
      val io = new Bundle {
        val out = UInt(OUTPUT, width=8)
      }

      io.out := UInt(1)
    }

sdtwigg, 2016-08-12:
This needs to be private[Chisel] :
https://github.com/ucb-bar/chisel3/blob/master/chiselFrontend/src/main/scala/chisel3/core/Data.scala#L69

zhemao, 2016-08-12:
Yeah, Steven is right. It's because Bundle has a method called width, so when you use "width" to declare the UInt, it uses that method instead of the parameter passed in.

aswaterman, 2016-08-12:
"Needs?"  I disagree, unless we provide an isWidthKnown method instead.

sdtwigg, 2016-08-12:
That would easily be best because that width returns a Width which is defined in:
https://github.com/ucb-bar/chisel3/blob/bc3813ebd752646a92987c74136bb2adb476b687/chiselFrontend/src/main/scala/chisel3/internal/firrtl/IR.scala#L106

I said 'needs' because I was under the impression that internal package means people shouldn't be directly manipulating those classes. (Otherwise, i should be moved out of the folder and package marked internal.)

aswaterman, 2016-08-12:
Yeah, I agree if the type Width is meant to be internal, the method width should be private.

PR #257 (2016-08-16): Make "def width" a private API; expose isWidthKnown instead
Resolves #256.

Since width was used to determine whether getWidth would succeed, I added

    def isWidthKnown: Boolean

but another option would be to expose something like

    def widthOption: Option[Int]

...thoughts?
commit ddb72787 (2016-08-16): Make "def width" a private API; expose isWidthKnown instead (#257)

* Make "def width" a private API; expose isWidthKnown instead

Resolves #256.

Since width was used to determine whether getWidth would succeed, I added

    def isWidthKnown: Boolean

but another option would be to expose something like

    def widthOption: Option[Int]

...thoughts?

* Document getWidth/isWidthKnown

* Add widthOption for more idiomatic Scala manipulation of widths
closed 2016-08-16 by aswaterman