chipsalliance/chisel #260: More Vec woes: how to build a Vec of Vecs?

ben-k, 2016-08-17:
Judging from the number of open issues on it, it seems like Vec() is in a pretty sorry state in Chisel3 at the moment.  At the risk of piling on, the following snippet (the intent of which is to generate a Vec of Vecs with differing lengths) produces invalid firrtl:

    val seqOfInts = Seq(1,2,3)
    val VecOfVecs = Wire(Vec(seqOfInts.map(x => (Vec(x,UInt(width=32))))))

I won't pretend to know enough to propose a fix for all of these issues, but I will note that in the last few days I have inadvertently written code using Vecs that:

* Silently produces invalid firrtl
* Silently produces code that the firrtl compiler does not complain about, but results in verilog that is clearly nonsensical or differs from the written intent
* Silently produces code that causes the firrtl compiler to (silently) emit "poison verilog" that spins infinitely without progressing in simulation

It seems like there are too many constructors and not enough checking by the Chisel3 compiler that they are being used safely.

ducky64, 2017-02-06:
This should work? Otherwise superseded by #445

ducky64, 2017-02-06:
Also, Vecs must be homogeneous, otherwise use a Seq or Record.

closed 2017-02-06 by ducky64